PiSupply/PiJuice #672: Need help finding bp7x 3500mAh or higher alternative? (non external battery)

Jamos1988, 2021-03-10:
I am trying to find an alternative to the default bp7x "internal" battery that fits directly into the Pijuice's hat.
I need one with at least 3500 mAh, but I cannot have it as an external as I need it positioned in exactly the same way.
This is due to a custom pcb and casing that we are designing, so preferably a bettery that fits just as we have it now.
The trouble is, 1820 mAh is the biggest bp7x that I can find and nothing that has larger capacity. 
The battery may be higher than the original, but it must fit the dimensions so it fits in the hat.

Are there any other search terms or key words I can try? 
I know it is the battery that fits a Motorola Droid 2, but the biggest batteries for that are also 1820mAh.

I hope someone can point me in the right direction.

shawaj, 2021-03-13:
@Jamos1988 you can get 2300mAh ones like this https://www.amazon.co.uk/dp/B004HZ1QRK/ref=cm_sw_r_cp_apa_fabc_B62ZZAZH0GR7G0G1XBHP

I did once see a 3500mah one but can't seem to find it now.

shawaj, 2021-03-13:
@Jamos1988 this was the one but doesn't seem to be available anymore https://www.walmart.com/ip/YN4L-2-X-3500mAh-Extended-Batteries-for-Motorola-Droid-2-Milestone-A955-A855-A956-with-Black-Extended-Back-Cover/129204026

Having said that, how many do you need? As I might have one of those in my drawers at the office somewhere potentially.

Jamos1988, 2021-03-15:
Oh nice! That is a great help! I'll look into it.

Jamos1988, 2021-03-15:
Doesn't seem that the 3500mah one is available anywhere unfortunately. At the moment I just need one for testing, but if all goes well I will need it mass produced into our product. If it is not available I need to integrated the external batteries from Pjuice, for example the 12000mah one, into a custom PCB or mold. That would take more effort. A standard battery format would be very nice in this case. Also because of the fact the wires have to be stripped first and then put into the terminal, and screw it secure. Just "clicking" a battery in would be so much less hassle.

Jamos1988, 2021-03-15:
By the way do you know if a 18650 battery is compatible? This would open up some possibilities.

shawaj, 2021-03-15:
Yes an 18650 cell, I believe, is a single cell LiPo. However I would advise to try and get one with the inbuilt protection circuitry.

Unfortunately as these phones are no longer made the batteries are difficult to get hold of. Especially the larger ones.

We are also working in #580 on having a 3 pin click in connector on the new PiJuice hardware to avoid having to strip wires etc. But unfortunately we don't have a timeline on that yet.

Jamos1988, 2021-03-16:
Yes, I can imagine sourcing the bp7x even the 1820mah one would be a challenge in the near future. Great to hear that indeed the 18650 is compatible, as it is a single cell LiPo. We will definitely get the protected ones. Anything else I should know, software firmware wise or any issues I might expect?

The connector would reduce assembly time and complexity enormously, so in our opinion sooner than later. 

Question: If all goes well we'd like to integrate this product into our device. Is there an option to speak to anyone about ordering larger quantities, and about any special requests? Things like; we need the dip switches to have a default setting, the i2c address of RTC to be different by default, and that they contain the newest firmware.

We have tried Sixfab ups, and Strompi by Joy-it. So we have come from quite a struggle to find the right product. Each with their own advantages / disadvantages, but in the end we are prefering PiJuice more and more. Just these last challenges to overcome.

Hope you are willing to discuss this further so we can go forward. 😄

shawaj, 2021-03-16:
@Jamos1988 for bulk orders please email sales@pi-supply.com

Re adding the battery, you would just need to set up the custom battery profile based on your battery specs.

Most of the config stuff you mention can be done automatically on the Pi side using the back up and restore settings stuff in the PiJuice utils

shawaj, 2021-03-16:
https://github.com/PiSupply/PiJuice/tree/master/Software/Source/Utilities