PiSupply/PiJuice #675: Press and hold sw1 is not sending "halt" signal to rpi?

petercwallis, 2021-03-14:
This is an old PiJuice kit I had in the cupboard (battery seems fine) and I have mounted it on a A+ running buster.
- power is provided via the PiJuice board
- The hard shutdown works (hold for 20sec)
- There is a suspect motor driver board (again found in cupboard) stacked on top of the PiJuice board 
- I did not add any pogo pins (reading the documentation it seemed I didn't need to; should I?)
- I am driving this from java so it takes a bit of effort to try different commands from software

P

tvoverbeek, 2021-03-14:
Since it was in the cupboard so long it probably has an old firmware.
Did you install the pijuice-base package?
The pijuice service needs to be running to get the shutdown after 10 seconds press of SW1.
Can you check your settings using the pijucie-cli?

petercwallis, 2021-03-15:
Yep. ran pijuice_cli and all looks good. Firmware is version 0.0 but Path is /usr/share/pijuice/data/firmware/PiJuice-V1.4_2020_01_17.elf.binary ..
Hmm...
I have the option to update it.  Will bravely push the button...
courage is failing with the message "are you sure"
Firmware update successful: V1.4
...
Nope. Same behaviour.
Rebooted the system (sudo shutdown .. + hard reset pijuice)
and find the firmware version is back to 0.0
Tried running pijuice_cli as root but it can't find I2C
Hmm..

tvoverbeek, 2021-03-15:
Smells like I2C is not working.
Have you installed i2c-tools?
If not, please do, and look at the output of `i2cdetect -y 1`.
You should see entries for address 0x14 (the pijuice itsel) and 0x68 (the RTC in the pijuice microcontroller).

petercwallis, 2021-03-15:
Done.  yes both as root and as pi I get 0x14 and 0x68, and 0x19 and 0x1e (being the compass/accelerometer attached)
Should pijuice_cli need running as root?  If not would I have odd permissions on some file on the pi somewhere perhaps?

tvoverbeek, 2021-03-15:
pijuice_cli is a set-uid wrapper for pijuice_cli.py and runs as user pijuice,
Also the service runs as user pijuice.
In pijuice-clli can you check the settings for button SW1 on the Buttons tab and report back.

petercwallis, 2021-03-15:
All three are empty:
< SW1 >
< SW2 >
< SW3 >

< Refresh >
< Back >

tvoverbeek, 2021-03-15:
You have to select <SW1> and press enter to get to the SW1 settings.
Here is what it looks like for me:
[image]
The long_press1 setting is the one for shutdown/halt.

petercwallis, 2021-03-15:
Ok and yes that is the values I have (now. was 10000 and 20000 for long_press 1&2 - assuming that should be fine.)

tvoverbeek, 2021-03-15:
10000 is 10000 msec = 10 sec
20000 is 20 sec before the hard power off.
The HARD_FUNC_POWER_OFF is done by the PiJuice firmware (no external software).
The SYS_FUNC_HALT needs the pijuice service to shut down the Pi.
What is the output from `systemctl status pijuice` (= the pijucie service service)?

petercwallis, 2021-03-15:
Ah, 
--
pijuice.service - PiJuice status service
  Loaded: loaded (/lib/systemd/system/pijuice.service; disabled; vendor preset: enabled)
  Active: inactive (dead)
--
I do not usually use python ... but /usr/bin/python is there and it runs...

petercwallis, 2021-03-15:
sorry about the format there :-/

petercwallis, 2021-03-15:
Ok so
 /var/lib/pijuice/
is there and owned by pijuice, but is neither readable or xecutable by group or other..

To run java as root I have put a '!' before "env_reset" in /etc/sudoers - can't think how that would affect pijuce running things owned by pijuice though..

petercwallis, 2021-03-15:
Okay I have enabled the pijuice service using
>sudo systemctl enable pijuice.service
and the button now successfully shuts down the pi after the specified delay.  Yay! I guess the need to do that is somewhere in the documentation - not that I didn't look.

However.  The soft shutdown does not turn off power to the pi?

tvoverbeek, 2021-03-15:
If you want to turn off the power you have to change the function for long press1 on SW1 from SYS_FUNC_HALT to SYS_FUNC_HALT_POW_OFF. That will turn off the 5V from the PiJuice to the RPi 30 seconds after the shutdon is initiated.

Note the last line in /etc/sudoers:  `#includedir /etc/sudoers.d`
The file `020_pijuice-nopasswd` in /etc/sudoers.d contains the necessary settings for the pijuice user.