PiSupply/PiJuice #676: PiJuice RTC Not Updating Due to Daylight Saving Time Change

kneiser, 2021-03-15:
This weekend daylight saving time changed in our geographic area. 

The raspberry pi system time is correct: date
**Mon 15 Mar 2021 09:32:51 AM MDT**

But the pijuice RTC is 1 hour behind: cat /proc/driver/rtc

**rtc_time	: 08:33:35**
rtc_date	: 2021-03-15
alrm_time	: 14:40:07
alrm_date	: 2021-03-15
alarm_IRQ	: no
alrm_pending	: no
update IRQ enabled	: no
periodic IRQ enabled	: no
periodic IRQ frequency	: 1
max user IRQ frequency	: 64
24hr		: yes

Based on pijuice documentation and other issues posted I am aware the the RTC gets periodically updated. How does one ensure that the pijuice supports the daylight saving changes?

tvoverbeek, 2021-03-15:
The RTC should normally be kept in UTC, not in local time.
When Raspbian updates the RTC it does it in UTC.
Also the system clock in  Raspbian/Linux is kept in UTC.
UTC has no summer time.
It seems the conversion to local time in /proc/driver/rtc is not correct.
Maybe it only updates the offset from UTC at boot.

Does `sudo hwclock -r -v` show the correct local time?
It also shows the 'hw clock time' which should be UTC.

kneiser, 2021-03-15:
Raspbian Clock: **date**
Mon 15 Mar 2021 01:32:07 PM MDT
**sudo hwclock -r -v:**

hwclock from util-linux 2.33.1
System Time: 1615836730.169782
Trying to open: /dev/rtc0
Using the rtc interface to the clock.
Last drift adjustment done at 1615822694 seconds after 1969
Last calibration done at 1615822694 seconds after 1969
Hardware clock is on local time
Assuming hardware clock is kept in local time.
Waiting for clock tick...
ioctl(4, RTC_UIE_ON, 0): Invalid argument
Waiting in loop for time from /dev/rtc0 to change
...got clock tick
**Time read from Hardware Clock: 2021/03/15 12:32:11**
Hw clock time : 2021/03/15 12:32:11 = 1615833131 seconds since 1969
Time since last adjustment is 10437 seconds
Calculated Hardware Clock drift is 0.000000 seconds
2021-03-15 12:32:10.641927-06:00

kneiser, 2021-03-15:
I set the system clock back to UTC: **sudo timedatectl set-local-rtc 0**
Here is the status: **timedatectl stat**us
               **Local time: Mon 2021-03-15 13:54:58 MDT
           Universal time: Mon 2021-03-15 19:54:58 UTC**
                 RTC time: Mon 2021-03-15 19:54:58
                Time zone: America/Edmonton (MDT, -0600)
System clock synchronized: yes
              NTP service: inactive
          **RTC in local TZ: no**
System time: **date**
Mon 15 Mar 2021 01:55:08 PM MDT

pijuice_cli:
[image]

I wanted the system to boot at 14:00, it did not. Everything was working on local RTC time until the daylight savings changed.

kneiser, 2021-03-15:
So I did one more experiment. I change the alarm to a time shortly after the current UTC time displayed in the pijuice_cli. It booted correctly.
So this means that every time there is s daylight savings time change that I will have to reschedule the alarm? As it will be out by one hours for each daylight savings time change.

Has anyone thought of an ingenious way to make this less confusing for use pijuice owners? Like an update to the pijuice software/firmware?

tvoverbeek, 2021-03-16:
If you let the OS update the RTC via timedatectl or timesyncd the RTC time is in UTC.
For MST UTC is 7 hours behind UTC and for MDT 6 hours behind UTC.
This fits with your timedatectl output:
```
Local time: Mon 2021-03-15 13:54:58 MDT
Universal time: Mon 2021-03-15 19:54:58 UTC
RTC time: Mon 2021-03-15 19:54:58
```

Above your hardware clock is on local time.
The RTC has no knowledge about timezones and when changes occur from summer toi standard time and vice versa.
If you want to keep your hardware clock in local time, you have to take care of the time changes.

kneiser, 2021-03-16:
I will be leaving the RTC to be updated by the Raspbian time syncing process. Perhaps the above information from this issue could be summarized in the Software section somewhere?

closed 2021-03-16 by kneiser