PiSupply/PiJuice #677: I2C documentation

petercwallis, 2021-03-15:
I am using the pijuice hat with java and need to write my own interface.
Working through the code at:
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py
I can find the I2C codes to send, but the results can be a bit .. interesting. I suspect that the problem is that, not being an expert  python person, I cannot tell the size of things to send, and indeed the documentation I can find on the parity check is fairly thin. Is there more documentation somewhere? If not, perhaps these questions can be answered.
- is there a parity check?
- When writing, is it always 2 bytes of data?  If so, how does the parity work?

tvoverbeek, 2021-03-15:
I2C is not like serial. It has no parity.
The pijuice.py code is still the best available documentation for the low level protocol.
The writes and reads are variable in length. On writes a checksum is added (exclusive-or of the message bytes).
The firmware adds the same checksum on reads, so you can check for correct reception.

shawaj, 2021-03-15:
There was some potential work suggested in #574 by @DazFahy and @mmilann to update the docs to add more useful info on this but unfortunately we haven't got around to it yet. I definitely think it would be great to have though.

petercwallis, 2021-03-16:
Okay. But I cannot tell from the python whether the board is expecting the delay argument to the PowerOff() function call to be 2 bytes long (a one bye int plus its checksum), 4 bytes, or 8 bytes for a four byte int and its checksum. The python presumably just sends whatever it sends; the firmware presumably has been written to match whatever the python sends.  And I need to match the firmware.  All that is really needed is someone to go through the firmware an make a list of the *_CMD values, the number of bytes expected, what is returned, and a bit of a description of what those bytes mean.  Boring but not a lot of work :-)

I will experiment and hope I don't break anything...

petercwallis, 2021-03-16:
Okay! experimenting didn't seem to break anything :-)
using pigpio.i2cWriteI2CBlockData(handle, POWER_OFF_CMD, data, 2); works.
That is the power off delay is represented by a single byte and the second byte is the checksum

I am closing this because I think we know what to do in future.

closed 2021-03-16 by petercwallis