PiSupply/PiJuice #679: Submit v1.7 debs to RasPi repos

shawaj, 2021-03-15:
Submit v1.7 debs to RasPi repos

Follows on from #612 and #678

shawaj, 2021-03-15:
@XECDesign just FYI, have submitted these debs to raspi repos now :-)
[Archive 3.zip](https://github.com/PiSupply/PiJuice/files/6143633/Archive.3.zip)

shawaj, 2021-03-15:
cc @mmilann @tvoverbeek

mmilann, 2021-03-16:
@shawaj Thanks, will add missing description about new features.

shawaj, 2021-03-16:
@mmilann you can see issue about that in #680 

If you want, you can just add rough overview and details about all the new features to that issue and I can add to readme if it is easier

shawaj, 2021-03-16:
Serge replied saying:

```
Hi Aaron,

Thanks for the update. We have updated the system a bit and it's
currently struggling with these uploads. It looks like the .sig file
isn't a detached signature, so it's not suitable for verifying the .deb
file?

I've added the files manually, but it would've gone through
automatically if the verification worked.

Kind regards
```

So looks like these should be in the repos now but I'll see if I can sort out the signature issues.

closed 2021-03-16 by shawaj

tvoverbeek, 2021-03-16:
Just checked, they are in the repos.
Noted they are also in the 64-bit version, which is not going to work since the pijuice_cli and pijuice_gui wrappers are 32-bit only.

shawaj, 2021-03-17:
@tvoverbeek let me speak to Serge.

I guess we should look to support the 64 bit versions if we can. Will start a new issue though

Jamos1988, 2021-04-07:
Hi, I have the newest 1.7 version of the pijuice GUI and the 1.5 firmware. The 12000mah battery still reports -9 Celsius and thus doesn't charge. How can I override the NTC resistor value, and to which value must I change? The link to custom battery profile in https://github.com/PiSupply/PiJuice/issues/659 is not visible anymore: https://github.com/PiSupply/PiJuice/blob/v1.7/Software/Configs/Batteries/LIPO_12000_PIS-1129.JSON

For now, I have shutdown the temperature sense and it charges, but I'd rather have the temp sense. 
I thought by upgrading to newest GUI and Firmware this would be solved, but maybe I am missing something?

tvoverbeek, 2021-04-07:
Current link is https://github.com/PiSupply/PiJuice/blob/master/Software/Configs/Batteries/LIPO_12000_PIS-1129.JSON

Jamos1988, 2021-04-07:
Thank you, I've applied the values in the custom profile. Temp is now -7 -8 sometimes, instead of a constant -9. So something seems to be happening, but still the wrong values. Anything I'm missing?

tvoverbeek, 2021-04-07:
What happens with the original LIPO_1200 profile?