PiSupply/PiJuice #680: Update docs for v1.7 software and v1.5 firmware

shawaj, 2021-03-15:
Update docs in relation to #612 and #678

shawaj, 2021-03-30:
[PiJuice Software V1.7 new features.pdf](https://github.com/PiSupply/PiJuice/files/6229376/PiJuice.Software.V1.7.new.features.pdf)

New features doc

shawaj, 2021-03-30:
@ChristopherRush if you get a chance, could you look at adding the stuff in above PDF to the documentation?

I guess we also need to add details about the new battery profile and the ability to override the NTC resistance parameter too.

@mmilann @tvoverbeek anything else to add?

mmilann, 2021-03-30:
Yes Custom profile for 12000mAh could be added to documentation, link: https://github.com/PiSupply/PiJuice/commit/d7940f95edb7ac28c419b4118b2607f582038cda