PiSupply/PiJuice #681: problem to operate rtc from pijuice

rascalos, 2021-03-16:
Hello,
I have to deploy some Raspberry PI4 with Pijuice.
I duplicate a sdcard from a functional equipment but we meet problem with rtc when we use copied sdcard on other Raspberry PI4 with Pijuice.
At the beginning, i2cdetect command showed us 68 at address 68 instead 'UU'. Problem solved by your documentation by adding dtoverlay=i2c-rtc,ds1339 in /boot/config.txt.
But always we meet problem with commands : hwclock -r or -s a and pijuice_cli crashes with some options.
Could you help us to solve theses problems, please ?
Regards

rascalos, 2021-03-16:
additional information, pijuice use V1.3 firmware. Packages pijuice_cli and pijuice_base V1.5 installed.

rascalos, 2021-03-16:
sorry pijuice_base and pijuice_guy V1.5 installed

tvoverbeek, 2021-03-16:
First the V1.5 pijuice-base package contains the V1.4 firmware. Please update the firmware.
Question: When do you execute `hwclock -r`? If it is at boot it might be too early before the RTC in the PiJuice can respond to I2C commands.

rascalos, 2021-03-16:
The last file of pijuice-base package is /usr/share/pijuice/data/firmware/PiJuice-V1.3_2019_02_28.elf.binary. I thought I was in V1.3
in addition the pijuice is in V1.3.
Concerning hwclock -r, I run it after the boot is finished.
On the other hand, the hwclock -s command is the first to run because in the /etc/rc.local file.
Ok I am going to update and come back to you.
Thanks for your help.

rascalos, 2021-03-17:
Hello,
Thanks for your advices. After updating the firmware it works.
Regards,

closed 2021-03-17 by rascalos