PiSupply/PiJuice #687: COMMUNICATION_ERROR

ncklamov, 2021-03-18:
Hi

I keep on getting ‘COMMUNICATION_ERROR’ every 30 seconds to a minute for about 10 seconds before resuming back to normal.

I am currently powering the Pijuice through the Raspberry Pi rather than the Pijuice because I read it can be powered from either ports. I have tried the micro-usb port too. Power source that is incoming is 5.1v stable.

Out of the box the battery it came with ‘BP7X_1820’ is also not connecting to the Pijuice. It looks perfectly fine as well as the pins on the PCB of the Pijuice - but it still says under battery status NOT_PRESENT.

I have also attached a photo of the information on the GUI when normal.

I am more interested in fixing the stability problem rather than the battery because I do not need the functionality of the battery, as I only need the RTC to work and of course be stable.

I have verified the Pijuice is correctly connected to my Raspberry Pi.

ncklamov, 2021-03-18:
[image]

shawaj, 2021-03-18:
@ncklamov can you make sure the software is the latest v1.7 version by doing an apt update and apt upgrade in the terminal?

Then can you go to "Configure HAT" menu and then the firmware update tab and make sure you are up to date?

With the battery not present, did you definitely remove the plastic safety tab between the battery and the PiJuice?

jahau, 2021-04-25:
After upgrading PiJuice firmware to 1.5 I haven't seen these errors any more.