PiSupply/PiJuice #691: After powerfailure blue no lights come on

sandercox, 2021-03-23:
I have a PiJuice attached to a RPi 4 - 4GB.
My power supply, USB-C, is connected to the Raspberry Pi and at first everything worked fine. Have this setup with Home Assistant and basically use it so I can pair Z-Wave modules around the house without having to restart all the time.

Now after a power outage that was longer than the piJuice could handle my pi started up but no lights started blinking on the pijuice. Rebooted everything, no change. Removed battery and re insterted it - no change. No matter what I do I don't see any changes.

Unfortunately there are no native connections for getting data from PiJuice in HomeAssistant but I've been experimenting a bit and now have data from pijuice_util script:

```
PiJuice UTIL: get time
{'second': 33, 'minute': 37, 'hour': 18, 'weekday': 6, 'day': 6, 'month': 1, 'year': 2000, 'subsecond': 0, 'daylightsaving': 'NONE', 'storeoperation': False}

PiJuice UTIL: get alarm
Alarm:  {'second': 0, 'minute': 0, 'hour': 0, 'day': 0}
Controlstatus:  {'alarm_wakeup_enabled': False, 'alarm_flag': False}

PiJuice UTIL: get status
{'data': {'isFault': False, 'isButton': False, 'battery': 'NORMAL', 'powerInput': 'NOT_PRESENT', 'powerInput5vIo': 'PRESENT'}, 'error': 'NO_ERROR'}

PiJuice UTIL: get config
ChargingConfig:  {'charging_enabled': True}
Battery Profile:  {'capacity': 1000, 'chargeCurrent': 625, 'terminationCurrent': 50, 'regulationVoltage': 4180, 'cutoffVoltage': 3000, 'tempCold': 0, 'tempCool': 2, 'tempWarm': 49, 'tempHot': 65, 'ntcB': 3450, 'ntcResistance': 10000}
Firmware Version:  {'version': '1.4', 'variant': '0'}

PiJuice UTIL: get battery
{'batteryCurrent': 4, 'batteryVoltage': 4111, 'chargeLevel': 93, 'batteryStatus': 'NORMAL'}

PiJuice UTIL: get input
{'ioVoltage': 5191, 'ioCurrent': -596, 'gpioPowerStatus': 'PRESENT', 'usbPowerInput': 'NOT_PRESENT'}
```

So the HAT seems to be functioning the battery seems to be charged (if I interpret results correctly) but as soon as I disconnect my USB-C power my Pi is shutdown. So no power is flowing from the PiJuice into the PI.

What can I do to get this restored?

mmilann, 2021-03-23:
@sandercox try to reset pijuice configuration to default from general menu tab.

sandercox, 2021-03-23:
As this is on a Home Assistant config I do not have a direct terminal / GUI on this Raspberry PI. I've come so far as to create some docker AddOn in HomeAssistant from when I can launch CLI commands but no menu navigation or anything is possible.

Any method to do this as a CLI command?

mmilann, 2021-03-23:
You can reset configuration from pijuice_cli general menu, or you can do it just by dual long press of SW1 and SW2 for 20 seconds

sandercox, 2021-03-23:
Managed to get there thru some hoops. Still no light on D1 but I managed to get D2 to work and now power loss no longer stops my pi. Anything I can try to get D1 to operate as well too?

mmilann, 2021-03-24:
You can configure from pijuice_cli ->LEDs menu one of LEDs to show battery charge status, another one can be configured as user led so you can send LED control commands from software.