PiSupply/PiJuice #694: Batt temp stuck at 46ºC since update from SW1.6/FW1.4 to SW1.7/FW1.5

Patag, 2021-04-02:
Hi all,
All is in the title I hope. This occured right after the SW/FW update. 
Pijuice GUI, pijuice_cli and even external readings via API show the same result.
I have now to tune "warm" threshold to 47ºC to allow battery charging.
My conf : RPi 4 / Raspbian 32bit / Pijuice with SNN5843 battery (2300 mAh).
Did I miss something during update ?
Thanks in advance.

alpapak, 2021-04-02:
I have the same issue. Try to blow some air between pijuice and pi.

Patag, 2021-04-02:
Thanks @alpapak. But a fan is already installed in my system ( directly driven by Raspbian OS).
The temperature measurement was running fine before SW/FW update.

tvoverbeek, 2021-04-02:
What is your Temperature sense setting AUTO_DETECT, NTC, ON_BOARD or NOT_USED?
If you use NOT_USED the firmware assumes 25 degC.
Check what you get when using NTC and ON_BOARD.

Patag, 2021-04-03:
Hi @tvoverbeek ,
Thanks for your reply. After playing arround with temp sense parameter, it seems something has been "unlocked". NOT_USED shows 25°C whereas all other show >= 46°C (47, 48, no more stuck to 46). Back to AUTO_DETECT, I get now a very strange behavior : here is a log generated by my Home automation system.
[image]
Weird no ?

alpapak, 2021-04-03:
@Patag Could you please post a photo of your raspberry with the fan and pijuice.

Patag, 2021-04-04:
Hi @alpapak 
I'm using this enclosure from Geekworm
[image]
Sorry, but I've to shut down my system before opening case due to the Usb3 extermal SSD connector. Next time I'll have to open it, I'll send you pictures for sure.
[image]