PiSupply/PiJuice #699: PiJuice cutting off raspberry pi’s power

priyankapote96, 2021-04-06:
I have connected PiJuice Zero to Raspberry pi 4 model that comes with 4 GB ram, also I have connected SSD via USB 2.0 and my custom USB powered sensor circuit via USB 3.0 to raspberry pi
My custom USB powered sensor circuit requires maximum 400mA current only 
There are 10 lithium-ion battery cells (each 3.4V and 2000mah capacity) in parallel to power my Raspberry Pi through PiJuice Zero
I have calibrated the above customised battery according to the PiJuice documentation and used the calibrated parameters in the configure HAT> Batttery settings and also Selected DIRECT_BY_MCU option for RSoC estimation as shown in the link below:

https://drive.google.com/file/d/1Q74tnQssKZAdTCDDCZK4nZxYlfidg4Rx/view?usp=sharing 

I have done buttons setting as shown in the link below:

https://drive.google.com/file/d/1cMZDKBmzwPNverqc-g_upz9GK2ZvNgIA/view?usp=sharing

The issue that I am facing is, my pijuice is cutting off the power to my raspberry pi even though battery is fully charged. When I turn on raspberry pi using SW1 of PiJuice the raspberry starts and then turns off immediately and all raspberry pi’s glowing LEDs also turn off immediately. If I try to turn on raspberry pi again using the same SW1 of PiJuice same way again my raspberry pi turns off in a few seconds
When I connect an external power source such as an external power bank to the pijuice, to charge the battery and then turn on raspberry pi using SW1 of the PiJuice then raspberry pi boots up successfully and works for longer duration. Now when I remove the power supply again, the same thing happens and the raspberry pi turns off. 
In a batch, 50% of my units exhibit this problem. This is not common for all. Also this issue begins to appear after some days of usage.

tvoverbeek, 2021-04-06:
Sorry, not able to help you. The PiJuice is for a single cell LiPo battery only, not for 10 in parallel. I do not think it will be able to deliver the required charging current.

priyankapote96, 2021-04-06:
> Sorry, not able to help you. The PiJuice is for a single cell LiPo battery only, not for 10 in parallel. I do not think it will be able to deliver the required charging current.

In my case pijuice is still able to charge the battery and run raspberry pi, the above mentioned problem appear when i am not charging battery and its fully charged and i turn the raspberry pi 4 on using pijuice and battery is connected to pijuice

priyankapote96, 2021-04-12:
> Sorry, not able to help you. The PiJuice is for a single cell LiPo battery only, not for 10 in parallel. I do not think it will be able to deliver the required charging current.

Hello Ton , 

Thank you very much for information, it helped me understand that the battery change is required

I have changed the battery to 10000mAh single cell LiPo battery and now PiJuice stopped cutting off the power to Raspberry pi from battery. Changing the battery to single cell LiPo battery helped to solved the power issue.
but now also whenever i put this single cell LiPo battery on charging through PiJuice, it shows weak indication at USB micro power input even when battery is already 50 percent charged 

initially when i connected this 1000mAh LiPo battery to PiJuice and did put on charging that time it was showing Present but after one day of usage of the battery ,now whenever i put battery on charging through PiJuice it show weak indication

should i really consider weak indication being a problem ?

tvoverbeek, 2021-04-12:
Input power weak means there is insufficient power to both charge the battery and keep the Pi running. Remember especially the Pi 4B is more power hungry than previous models.

priyankapote96, 2021-04-24:
hello, thank you so much for explaining weak signal of pijuice 

Also now i have used LiPo single cell battery which is 10000mAh battery and its not from pijuice pi supply, i have baught brand new 10000mAh LiPo single cell battery from market. i am facing the issue with battery charging state 

i have used the PJLIPO12000 pijuice battery default profile from configuration and changed the battery charge to 10000  as shown below:
https://drive.google.com/file/d/14iW6FsKmsbhD2pc3GYJgFxWT_2mYYEre/view?usp=sharing

i am having issues with the battery charge percentage shown by the pijuice, pijuice shows battery percentage from 89 to 14 just in one second even though i have charged my battery for 6 hours thoruhg pijuice and also sometimes shos charge percentage from 9 to 50 or 60 even when i dont charge my battery , and its been happening since a week and i have used brand new battery and pijuice , i have 5 pijuice for testing and all the pijuice are showing similar behaviour that it shows high to low charge percentage within a blink of an eye even after charging battery for longer hours and sometimes it shows from low to high charge percentage even without charging a battery 

i am not understanding where i ahve done mistake

tvoverbeek, 2021-04-26:
PiJuice as delivered might still have the previous firmware (1.4).
Make sure you update the pijuoce-base and pijuice-gui packages to 1.7 and the firmware to 1.5.
See if you then still have the problem.

priyankapote96, 2021-04-27:
hi, thank you for information

I have found the root cause of the problem that i am facing with my battery percentage dropping suddenly and it happens only when the pijuice shows communication error 

but this communication error is not continues it appears once or twice in an hour but this error goes automatically leaving my battery percentage at wrong stage 

so the main issue is this communication error of pijuice nd what should i do for this ?

i tried fixing the core frequency to 500 and minimum core frequency to 500 but still it is not solving any communication error that is appearing anytime and goes away and disrupting my battery charge level 

how to fix this error , is there nay way to set i2c frequency of pijuice to fix speed?

tvoverbeek, 2021-04-27:
Again, what version of the firmware are you running? 1.4 definitely has problems on the Pi4. Even with 1.5 you may need the core_freq and core_freq_min especially if you have a varying cpu load which varies the core frequency.

priyankapote96, 2021-04-28:
hi, 
i have updated the firmware of pijuice to 1.5 and it shows successfully upadted to 1.5 version and if i clos all the pijuice gui window and go back again to check if firmware version is updated or not it show that firmware version is 1.4 and update is available for 1.5, i have tried this multiple times but still it is not upadting on gui from 1.4 to 1.5

tvoverbeek, 2021-04-28:
How did you update? If with the gui, did you get the progresbar  during the update. Also at the end of the update the firmware restarts. Did the PiJuice led come back on? (it is off during the update)

priyankapote96, 2021-04-28:
yes firmware restarted after the upadted and during the update there was communication error shown on gui and progress bar appeared 
when i upadted the software the below wwas shown on gui:
https://drive.google.com/file/d/1ZqWpa7FqGdQergKXPgeHYOLsb1ZNAF0d/view?usp=sharing

when i closed the gui and checked the firmware section again it was showing below:
https://drive.google.com/file/d/1AVeOOcmNfNREB2uskYdgFYjXxO-uhl0P/view?usp=sharing
 and also yes the pijuice led come back on

tvoverbeek, 2021-04-28:
Communication error during update is no good.
Try to update directly in a terminal window:
`pijuiceboot 14 /usr/share/pijuice/data/firmware/PiJuice-V1.5_2021_02_06.elf.binary`
See if you get any error messages which can help us further.

priyankapote96, 2021-04-29:
hi,
I have tried this method also with pijuiceboot to upgrade the firmware but still after the upgrade it still shows firmware version 1.4 when i check in firmware section of GUI

after the firmware update using pijuiceboot 14 , i shutdown the pi and disconnect the battery from pijuice and start everything fresh,but still firmware version section shows 1.4 instead of 1.5 

as you rightly said when i am doing firmware update uisng pijuiceboot 14 _filepath_ there is not communication error shown on gui that means the firmware update is successfully done but again still its not updated as 1.5 in firmware section of gui

tvoverbeek, 2021-04-29:
I am starting to suspect you have the wrong or a corrupted firmware file.
What is the size of your firmware file? 
Correct size for 1.5 is 77708.
For 1.4 it is 75156.
SHA-1 sums are:
```
b12ac7af3482905c9adbad96db3bba17882da2b6  PiJuice-V1.4_2020_01_17.elf.binary
459c8e395d664ee8fbcf01929e9e4633d4caba52  PiJuice-V1.5_2021_02_06.elf.binary
```
Please check if your 1.5 firmare file has the correct size and SHA-1 checksum (use sha1sum command on RPi)

priyankapote96, 2021-04-29:
yes, its working...........its showing now that firmware is up to date 1.5

i hope now that communication error issue also wont occur , its under testing now .....i will get back to you after testing if i incure any issue related to communication error

priyankapote96, 2021-07-15:
hi,

i am writing this issue after testing pijuice for almost after 2 months

i am still facing the issue where PiJuice is cutting off the power of my Rpi4 few min after the Rpi is booted, even the battery is at 85% shown by pijuice 

i am using lithium polymer 10000mAh single cell battery as internal battery to my device which is powering Rpi through PiJuice, and this setup is turning my Rpi4 off few mins after the boot up through J1 switch 

If is user external battery which is charging internal battery through PiJuice and also providing power to Rpi THrough PiJuice only, then only my device works for longer hours 

my device is a product which has Rpi and Pijuice as the core part, and internal battery is provided for portability, and beacause my device is turning off after few mins of boot up on the internal battery , the portability is gone and my clients have to use device on external battery only to keep Rpi4 runing for longer hours 

is it that my internla battery of 10000mAh is not charging through piJuice ?
or i am using wrong charging current in the pijuice setting ? as i am using 2500 charging current for 10000mAh battery 
is Pijuice showing wrong percentage of the battery RSOC and battery is not charged enough to provide power to Rpi?

what you think , need help as my clients are complaining and i will have to discard the idea of using PI and pi juice in my future products

tvoverbeek, 2021-07-16:
Do you know how much power your application is using (i.e. RPI4 + peripherals)?
My guess is the sum of this power and the power needed to charge the battery is more than the PiJuice can supply.

priyankapote96, 2021-07-22:
Hello Ton,

Finally the issue that i am facing is solved, and i am thanking you Ton for all the help and guidance 

your guess is right, it was not the problem related to pijuice or Rpi4 it was an issue related to peripheral that i have connected 

the data aquisition system that i have connected using USB to Rpi4 was consuming 400mA of current which was not an issue but the SSD that i have connected to Rpi4 was consuming 1amp of current for read and write operations , when i removed the SSD and used pendrive instead , my device was working aboslutely fine 

and the battery drainage has also slowed down now , i am happy and i can move forward with pijuice now with confidence

closed 2021-07-22 by priyankapote96