PiSupply/PiJuice #702: Battery percentage stuck on 0%

Callum6677, 2021-04-11:
I have a pijuice hat running software version 1.7 and firmware version 1.5

No matter what settings I change, the pijuice reports 0% battery at all times, freshly charged with external charger, fully depleted, storage charged. This is with RSoC estimation set to *either* AUTO_DETECT or DIRECT_BY_MCU
For reference this is the profile

{
'profile':{
    'capacity': 27000, 
    'chargeCurrent': 1975, 
    'terminationCurrent': 350, 
    'regulationVoltage': 3600, 
    'cutoffVoltage': 3000, 
    'tempCold': 0, 
    'tempCool': 10, 
    'tempWarm': 45, 
    'tempHot': 60, 
    'ntcB': 3977, 
    'ntcResistance': 10000
},
'extended':{
    'chemistry': 'LIFEPO4', 
    'ocv10': 3200, 
    'ocv50': 3300, 
    'ocv90': 3500, 
    'r10': 200, 
    'r50': 200, 
    'r90': 200
}
}

some of the ocv/r numbers are pretty much shot in the dark figures, but even so, changing them to incorrect values seemed to make zero difference, still flat 0% charge reporting

Callum6677, 2021-04-13:
Any assistance at all? Is this a bug? Maybe a config issue? I'm at a bit of a loss and it's not exactly useful having charge percentage stuck on 0%

tvoverbeek, 2021-04-13:
Are you sure it is LIFEPO4 battery and not a LIPO?
When using 'DIRECT_BY_MCU' it relies on correct values of ocvxx and rxx for the battery type.
What happens if you use NTC (if your battery has an NTC).
What is the indicated battery temperature? If it is over tempWarm charge current is reduced.
If above tempHot or below tempCold charging is suspended.

Callum6677, 2021-04-13:
It is definitely an LiFePO4 battery using decent quality cells which I can have been able to also measure capacity for via external chargers and verify accuracy against their advertised values. There is an NTC connected, indicated temperature across all testing has varied between 22 and 24c, which is in accordance with what I would expect considering room temperature and the large physical size of the battery pack given plenty of room to dissipate heat during charge/discharge with a raspberry pi, it is also neutral to touch.

The pijuice is charging the cell, it shows the correct voltage at all times. It just is stuck showing 0% capacity regardless of whether I set RSoC estimation to be based on AUTO_DETECT (which I believe uses the LiPo based fuel gauge) or DIRECT_BY_MCU which I was of the understanding is a much cruder algorithm based on voltage level, I do not have access to ocvxx and rxx values for these cells but would have assumed nice round estimated values would have shown *something* other than 0

tvoverbeek, 2021-04-13:
ocv10 means open charge voltage at 10% charge (battery without any load).
Have you connected the NTC lead to the battery plug? If not the PiJuice does not detect the NTC.
You wrote "It is definitely an LiFePO4 battery using decent quality cells" which means it is a multi-cell battery.
As stated in the documentation the PiJuice can only handle single-cell batteries.
You have to look for a different product it it is a multi-cell battery.

Callum6677, 2021-04-13:
It is a parallel configuration, only 3.2V "nominal" series voltage. This is a compatible configuration.

I am aware of what ocv10 means and aware of the fact the NTC needs to be connected, as stated, I have connected the NTC and the indicated temperature was already correct. Even if the NTC was in fact incorrect, I do not see how it would be relevant to this issue when the temperature *as measured by pijuice software* is within range.
It is however basic physics that the battery under load will never have a voltage that is *higher* than its open circuit voltage, therefore, you would expect that with the open circuit voltage for 10% specified to be 3200mV and a measured voltage of 3450mV (which I can verify with both a multimeter and an oscilloscope as being correct within an acceptable margin of error), the battery percentage must be *at least* 10%, and yet the pijuice displays 0.

tvoverbeek, 2021-04-13:
I am afraid I cannot help you any further.
You could look at the firmware source: https://github.com/PiSupply/PiJuice/tree/master/Firmware/Sources-V1.5_2021_02_06 especially the battery related files in the Src directory.
@mmilann Could you help?

mmilann, 2021-04-13:
Do you get correct rsoc measurement when you use standard battery shipped with pijuice and standard profile set and Rsoc estimation parameter set to DIRECT_BY_MCU?

Callum6677, 2021-04-17:
I can replicate the issue with the standard lipo and DIRECT_BY_MCU, however, I have found a workaround which in turn may give a slight hint as to what is going on.

If I pull the battery entirely, remove the hat from the pi and any power and leave it on my desk overnight, when I connect it the following morning it then works. It appears part of the problem may be the settings being ignored after being applied and needing a complete power cycle to actually be observed

jrhelbert, 2021-06-01:
I would expect that you ought to be seeing something other than 0%, but I think think both your OCV and r values are fairly incorrect and might be confusing the MCU. 

LiFePO4 batteries have a **very** flat discharge curve, and the majority of the discharge is at or above 3.2V. Setting your OCV10 to 3.2 can't be helping.

LiFePO4 batteries tend to have fairly low internal resistance values and at 27 Ah capacity, you should be nowhere near 200 mOhm (unless you have a _very_ poor connection between the PiJuice and the batteries. There is a test script in the repo here that you can use to characterize the batteries behavior by discharging into a test load and it will determine suitable OCVxx and Rxx values. At the super low IR that the LFP batteries use, you can't just the stated IR (if you even have it), you have to also factor in the cable losses to the battery as well.

For reference, I have two configs that I have characterized, one is 8x 32700 6000 mAh cells in parallel, here is the config I have for it, as reported by the aforementioned test script:
`{
'profile':{
'capacity': 48000,
'chargeCurrent': 2500,
'terminationCurrent': 400,
'regulationVoltage': 3500,
'cutoffVoltage': 2700,
'tempCold': 0,
'tempCool': 2,
'tempWarm': 49,
'tempHot': 65,
'ntcB': 3950,
'ntcResistance': 10000
},
'extended':{
'chemistry': 'LIFEPO4',
'ocv10': 3147,
'ocv50': 3240,
'ocv90': 3283,
'r10': 19.7,
'r50': 19.6,
'r90': 19.5
}
}`

I don't have the exact details for my other setup on hand, but its a single 90Ah prismatic and it's real Internal Resistance is even lower than those above, but the Rxx values are only 1 ohm or so less. I have short 14awg runs between the batteries and the PJ, and between that and the connectors I still see a significant adder, but for the MCU estimation to work, it has to know the resistance of the whole path between battery and PJ.

You might try substituting my OCVxx and Rxx values in quick to see if that helps at all before you bother trying to run the battery characterization script.

That said, the MCU estimation algorithm has **a lot** of problems when you start getting into very large capacity batteries. The voltage just doesn't change very fast and this is made even worse by the LFPs flat discharge curve. My setups work, but if i do a constant current discharge, the rSOC does not track the way that it should.