PiSupply/PiJuice #705: Significant battery drain during deep sleep

Guttagrynna, 2021-04-13:
Last week I received the PiJuice HAT I had ordered. But I have a problem with it. When I turn the RPi off and leave it without charging the battery drains in a couple of hours (less than ten hours). This can't be normal, can it? I have unplugged all extras, even the ethernet cable, reset software to defaults and the deep sleep mode is signalled by the LED. What am I missing? I don't want to return it if there's nothing wrong, obviously. But this will kill the battery prematurely and somewhat defeats the purpose of my purchasing it.

tvoverbeek, 2021-04-13:
How do you turn off the RPi? Is the red RPI power LED off?
It may be the Pijuice is still supplying 5V to the RPi. If any USB peripherals are connected they will then still draw current.
A good comparison can be to find out how long the RPi will run on a full charge (no external power).

Guttagrynna, 2021-04-13:
Thank you for your reply.

- I either turn off using SW1 for ten seconds or I VNC to the Pi and turn of using normal shutdown in the GUI.
- The red RPi power LED is on despite unplugging. I would have thought this does not necessarily imply excessive power consumption when turned off.
- The problem remains when I don't use any USB peripherals.
- I will experiment a bit woth power consumption during power on and off respectively.

tvoverbeek, 2021-04-13:
If you did not change the SW1 setting for the 10 sec press then the PiJuice keeps supplying 5V to the RPi.
In the GUI/CLI change the function of SW1 LONG_PRESS1 from SYS_FUNC_HALT to SYS_FUNC_HALT_POW_OFF.

For VNC shut down: on the System Task tab enable 'Software Halt Pow OFF' and set a time long enough for a proper shutdown e.g. 20 seconds.

Then the red led will go off after 30 seconds for SW1 and 20 seconds for shutdown via VNC.
The PiJuice battery will then keep its charge for a long time.

Guttagrynna, 2021-04-13:
That seems to work! At least the red LED goes off after 20 seconds. I will now leave it overnight, but I'm sure it will be fine. Thanks for your help!