PiSupply/PiJuice #708: Booting by attaching to power?

jahau, 2021-04-20:
Is it possible to configure PiJuice HAT so that it will boot (if shutdown of course) when the power supply is attached? In this way there wouldn't be need for an external button when PiJuice HAT buttons aren't available (=put into case making PiJuice HAT buttons unavailable). There's at least this "No power" scenario which is vice versa what is needed. Tanks!

tvoverbeek, 2021-04-20:
You can use the 'wakeup on charge' setting with a low level (0 or 1%).
When power is  connected charging will start nand the Pi will be woken up.
Use the latest SW version (1.7 and firmware 1.5) and enable storing the wakeup-on-charge setting.
Then the setting will be stored in the PiJuice non-volatile memory, otherwise you have to make sure the
setting is refreshed before shutting down.

jahau, 2021-04-20:
That was fast, thank you! I'll try it out.

jahau, 2021-04-20:
Okay, now I've tried but it doesn't work. I attached the power plug to Rasberry Pi (not Pijuice HAT since it's not visible in this scenario). Firmware is the latest one and in my understanding also software. la Should this still work?

tvoverbeek, 2021-04-20:
If you power via the Pi (and the Pi was off) it will boot immediately independent of the PiJuice. The 'wakeup-on-charge' only works if the Pi is powered via the PiJuice for obvious reasons.

jahau, 2021-04-21:
I didn't see this work like this, I had to push the button on PiJuice to boot the Pi. I'll re-check today since the sd-card got broken at some point.

jahau, 2021-04-25:
Doesn't work. I've upgraded firmware to 1.5. When shutting down Pi normally from the OS, then attaching power plug to Pi no boot will occur. Is there some setting on PiJuice to make this happen? Thanks.

tvoverbeek, 2021-04-25:
Before shutting down the Pi was on battery power? Did the PiJuice go to low power mode after the shutdown (slow weak blink)?

jahau, 2021-04-25:
I've tried both scenario, battery power and power plugged when shutting down Pi. In neither of the scenarios the Pi will boot when putting power plug back on...I guess this is just feature that it works like this.

octopusarm, 2021-05-20:
I've experienced an identical problem: when the pi is shutdown, the only way to boot it up again with the pijuice attached is to plug into the pijuice's microUSB port and push the SW1 button. Is this considered a normal behaviour? It makes things quite difficult, because in the case we are using, the pijuice microUSB is not accessible...

tvoverbeek, 2021-05-20:
Wake up on charge works.
Make sure you are using firmware 1.5 and software version (package version) 1.7.
When enabling wkae up oncharge also enable the 'restore' option. This stores the wakeup on charge
setting in the PiJuice non-volatile memory.
Before you had to make sure the wake up on charge setting was refreshed just before shutdown.

octopusarm, 2021-05-20:
> Wake up on charge works.
> Make sure you are using firmware 1.5 and software version (package version) 1.7.
> When enabling wkae up oncharge also enable the 'restore' option. This stores the wakeup on [...]

Thanks for the quick reply! I suspect the firmware might be the issue, as I've recently been setting up a large number of pijuices and have probably not updated all of them... I'll test the behaviour with updated firmware to see if it does what we need and edit this reply in case it's helpful for others...

mroelandts, 2024-09-04:
The "wakeup on charge" option works on firmware 1.6 in combination with the "restore".
But changing the trigger level to something else than the default value (50) is not stored apparently.
On the next boot this value is back to it's default value 50.
Is this expected?