PiSupply/PiJuice #709: Tray icon broken for new users.

git-tomb, 2021-04-22:
FYI, the tray icon only works for user pi.  It does not appear for any other user.

tvoverbeek, 2021-04-23:
More info please. See issue #390 were this was fixed.
Need more specific info about your scenario.

tvoverbeek, 2021-04-26:
@git-tomb Problem confirmed.
It is caused by the new init_pijuice_interface() function (needed because the PiJuice I2C address can now be changed on the fly).
The function needs read access to the JSON configuration file: `/var/lib/pijuice/pijuice_config.JSON`.
This file is only accessible for user pijuice.
User pi can start pijuice_tray.py as user pijuise because it has sudo power.
A normal new user cannot do this (sudo disabled by default).
Can be solved as follows:
As user pi (or root):
```
sudo chmod g+x /var/lib/pijuice
sudo chmod g+r /var/lib/pijuice/pijuice_config.JSON
```
Tested it and it works for me.

git-tomb, 2021-04-26:
That solution didn't work for me either. My new user is in the sudo group, and should be able to do everything pi can. Is it possible requiring a password for sudo with the new user is the cause?

tvoverbeek, 2021-04-26:
@git-tomb Probably. Look at the autostart desktop file /etc/xdg/autostart/pijuice-tray.desktop
where pijuice_tray.py is started with `sudo -u pijuice /usr/bin/pijuice_tray.py`.
Your new user in that case needs passwordless sudo as user pi.
See  `/etc/sudoers.d/010_pi-nopasswd` how this is done for user pi.

tvoverbeek, 2021-04-26:
@git-tomb It should also work for you.

Corrected instructions:
As user pi (or root):
`sudo chmod g+x /var/lib/pijuice`
`sudo chmod g+r /var/lib/pijuice/pijuice_config.JSON`
In /etc/xdg/autostart/pijuice-tray.desktop change the line
`Exec=sudo -u pijuice /usr/bin/pijuice_tray.py`
to
`Exec=/usr/bin/pijuice_tray.py`

Then it should work (finally)

git-tomb, 2021-04-26:
Created file /etc/sudoers.d/030_username-nopasswd containing the following:
username ALL=(ALL) PASSWD: ALL, NOPASSWD: /usr/bin/pijuice_tray.py

Allows tray to load without removing password protection for the entire system.

The problem will probably persist for users not in the sudo group, but that's above my pay grade.

Cheers.

git-tomb, 2021-04-26:
Removing sudo -u pijuice actually made it not work again>.<

Linux, amirite?