PiSupply/PiJuice #711: Wake up alarm not working after low charge shutdown  (but it turns on when battery reaches the treshold)

FernandaRamosTL, 2021-04-23:
Body:
Dear Ton:
I have several units of PJZero and Pijuice Hat (the big one).
Recently Iv purchased a new one, that came with a slightly different battery, and the new firmware version.
Something is happening:

The alarm is set to minutes period (turn on every 25 minutes). ALso, the minimum battery level is programmed to turno of the RPI when it is 50% and turn it back again when battery reaches the desired level of charge. (70%). (a remark: it keeps booting and shutting down before battery is 70%, and when it is fully charged, but the RPI is schedulle to shutdown on my crontab, it never wakes up again on the minutes period - see the image attached).

The period alarm was funcioning perfectly.
Then, the power was cut, and the battery started to drain, obviously. Then, the battery was again properly charged, and RPI turned back on again, as expectedly.

But, after this event, the alarm stoped working. This is very critical, since I cannot be present all times to manually turn the pi back again.

I have another unit programed exactly the same way, and this misbehaviour never happened. It has the older firmware. Both systems are connected to power using the Pijuice usb.

Can you help me with that? I just say that I cannot perform tests with that unit that had the problem because it is on the field. Luckly, when the problem happened, one of my clients employees was smart enouth to follow my instructions and turn the system back again, pressing the Pijuice hat button. But I absolutely cannot count with this every time. I do believe that this could be a firmware problem.

Thank you so much, be safe and take care,

Fernanda.

[image]

mmilann, 2021-04-26:
@FernandaRamosTL what are software and firmware versions?

FernandaRamosTL, 2021-04-26:
> @FernandaRamosTL what are software and firmware versions?

[image]

mmilann, 2021-04-26:
Hi @FernandaRamosTL There is newer software V1.7 and firmware V1.5 and advice is to update, there are some RTC issues fixed.

FernandaRamosTL, 2021-04-27:
> Hi @FernandaRamosTL There is newer software V1.7 and firmware V1.5 and advice is to update, there are some RTC issues fixed.

Thank youm @mmilann 
As the equipment is now on the field, I will wait about 2 weeks until I have it back to me, and perform the update :)
If you would be so kind to point me the directions to do so, I would really apreciate.
I have other several units of both pijuice zero and pijuice HAT, with older firmware versions. Should I update them? They are working just fine.

mmilann, 2021-04-27:
Hi @FernandaRamosTL , best to update to latest version. Process is simple, first update software:
`sudo apt-get update`
`sudo apt-get install pijuice-base`
and if you want GUI variant:
sudo apt-get install pijuice-gui

then open application (pijuice_cli from terminal in case of headless Os), go to Firmware tab and you can start update if not already latest version. All settings should be preserved after update.

FernandaRamosTL, 2021-04-27:
> Hi @FernandaRamosTL , best to update to latest version. Process is simple, first update software:
> `sudo apt-get update`
> `sudo apt-get install pijuice-base`
> and if you want GUI variant:
> sudo apt-get [...]

Thank you so much! I will try it with a system at my own place. :) After that, I will perform to the systems that are on the field, when I get it returned to my place :)

FernandaRamosTL, 2021-04-27:
> Hi @FernandaRamosTL There is newer software V1.7 and firmware V1.5 and advice is to update, there are some RTC issues fixed.

Hello again!
I updated the firmware and softare for the unit that is here in my house.
The behaviour changed:

Before, I hsd it set to shut down when power was cut. But as soon as I plugged the power back on, the unit would restart.

Now, when I cut the power off, the unit does not boot again.

Any ideas?

Thank you so much,

Fernanda.

tvoverbeek, 2021-04-27:
Assuming you apply external power to the PiJuice.
In the firmware there is now an option to store the wakeup-on-charge setting in non-volatile memory.
In the CLI or GUI on the System Task tab make sure both the wakeup-on-charge setting with a sufficient low threshold and the restore option are enabled.
Then the unit should restart when external power is reapplied.

FernandaRamosTL, 2021-04-27:
> Assuming you apply external power to the PiJuice.   _ YES, Im adoing it.

- "In the firmware there is now an option to store the wakeup-on-charge setting in non-volatile memory."  - HOW ?
- " In the CLI or GUI on the System Task tab make sure both the wakeup-on-charge setting with a sufficient low threshold and the restore option are enabled." - It is.
Can you enlight me :) ?
[image]
[image]
[image]
[image]
[image]

FernandaRamosTL, 2021-04-27:
> Assuming you apply external power to the PiJuice.
> In the firmware there is now an option to store the wakeup-on-charge setting in non-volatile memory.
> In the CLI or GUI on the System Task tab make [...]

Just more questions:
1) For the Pi turn on back again when power is plugged on, does it had to be previously shutdown because of power loss?
2) For the Pi turn on back again with power restabilished, does the battery be ABOVE the pre-determined low threshold?

Thank you!

tvoverbeek, 2021-04-28:
From the screenshots: You are still running the 1.6 version of the Raspbian packages. Please upgrade to 1.7 to see the Restore option. (`sudo apt update; sudo apt upgrade`) and ensure you have the 1.5 firmware.

For the questions:
`For the Pi turn on back again when power is plugged on, does it had to be previously shutdown because of power loss?`
No, with a user shutdown even without the removal of 5V to the Pi it works with the 1.5 firmare/1.7 software
`For the Pi turn on back again with power restabilished, does the battery be ABOVE the pre-determined low threshold?`
Yes, therefore set the threshold as lo as feasible for your application

FernandaRamosTL, 2021-04-28:
> From the screenshots: You are still running the 1.6 version of the Raspbian packages. Please upgrade to 1.7 to see the Restore option. (`sudo apt update; sudo apt upgrade`) and ensure you have the 1.5 firmware.

       **Dear Ton: I Had it updated first thing when dear Milan told me to do so, the first screen shot was 
      before the update. I send you the new screen shot. Anyhow, I am upgrading the Linux :)**

> For the questions:
> `For the Pi turn on back again when power is plugged on, does it had to be previously shutdown because of power loss?`
> No, with a user shutdown even without the removal of 5V to the Pi it works with the 1.5 firmare/1.7 software.

     **Dear Ton, I guess it is now working. I am running further tests just to be sure, and I am sending you new 
     screenshots.**

> `For the Pi turn on back again with power restabilished, does the battery be ABOVE the pre-determined low threshold?`
> Yes, therefore set the threshold as lo as feasible for your application   
       _**Dear Ton: I believe this was the reason why Pi was not turning on with power reaplied: The battery 
          was below the lower treshold, during my tests.**_

I will perform further tests today, and I will get back, probably with good news :)
[image]

FernandaRamosTL, 2021-04-28:
Hi again! I performed further tests. The pi will only reboot with power restabilished if the power was cut previously, according to the test that you can watch on the attached video.
I also attach more screenshots. I sudo upgraded and sudo updated.

Any thoughts?

[image]
[image]
[image]
![image](https://user-
[image]

images.githubusercontent.com/55566879/116405262-d5311a80-a805-11eb-8c6b-80b475bc60c3.png)
https://vimeo.com/manage/videos/542641008

FernandaRamosTL, 2021-04-28:
And, what this "restore " means? It was not there before.
[image]

FernandaRamosTL, 2021-04-28:
Dear Ton and dear Milan, I guess the problem is solved. I guess that the "restore" box was to be checked. Can you please confirm that this was the reason why PI was not turning on after manual shutdown with power restabilish? Thank we can close the issue :)

tvoverbeek, 2021-04-28:
Yes, before the Restore option was available you had to be sure wakeup-on-charge was recommanded before shutdown.
With Restore the setting is saved in NV memory (which is what I tried to point out in previous comments).
Closing the issue now.

closed 2021-04-28 by tvoverbeek

FernandaRamosTL, 2021-04-28:
> Yes, before the Restore option was available you had to be sure wakeup-on-charge was recommanded before shutdown.
> With Restore the setting is saved in NV memory (which is what I tried to point out in previous comments).

(I did not have noticed the small box before the tests, now it is done :)  Thank you :)))
> Closing the issue now.