PiSupply/PiJuice #715: Battery profile customization doubt

FernandaRamosTL, 2021-04-27:
Dear Tin and dear mmilan:
I am not sure if this is the right place for this question. If not, I apologise in advance.
I purchased a new battery to test with PIJUICE ZERO.

Question 1:
Can you help me to determine the proper values for the correct charging?

Question 2:

You mention that when using RPI 3, it is better using the Pijuice Hat then the Pijuice Zero.  What would be the implications? I purchased the battery that you can see at the picture so I will have more extra power for my Raspberry 3b.  It is a 3460 Mah 3,7volts litter battery.

The PJZero is more affordable for me, because I can purchase it using regular mail, since it does not contain any batteries (we pay 107% of taxes over imported products in Brazil, including the price of the mail, therefore a Pijuice HAT has a final cost, for me , of 157 pouds, and the pijuice zero has a final cost of about 50 pounds, so you can understand why PJzero is better for me).

Thank you in advance!

[image]

tvoverbeek, 2021-04-28:
For charge current use 0.5C which is here 1730 mA.
Leave Regulation Voltage and Cutoff voltage at their default values (4180 and 3000 mV).
Does your battery have a NTC (3-wires to the connector)?
If not, then put the Temperature Sense to 'not used'

I have read before about the extra high import duties in Brazil. I believe there is even a RPi version made in Brazil in a different color to avoid the extra duties (https://www.raspberrypi.org/blog/raspberry-pi-brazil/ from Nov 2017). 
How is the situation now in 2021?
Continue to use the PJZero. The PJZero has the same circuitry as the full HAT, except it has not HAT EEPROM, no dip switches and a different button configuration.

FernandaRamosTL, 2021-04-28:
Dear Ton, thank you very much!

> For charge current use 0.5C which is here 1730 mA.  **OK!**

> Leave Regulation Voltage and Cutoff voltage at their default values (4180 and 3000 mV). **OK**

> Does your battery have a NTC (3-wires to the connector)? **YES!!!**

> If not, then put the Temperature Sense to 'not used' - _ok, It does have the third wire._

**I will send you images until the end of this week of the battery itself and the battery configuration, just to be sure, if it is ok?**

> 
> I have read before about the extra high import duties in Brazil. I believe there is even a RPi version made in Brazil in a different color to avoid the extra duties (https://www.raspberrypi.org/blog/raspberry-pi-brazil/ from Nov 2017). - **Not really. The price is  about the same, and it is very difficult to find the brazilian version for sale. The only difference is that the brazilian version has "ANATEL" printed on it.** 

> How is the situation now in 2021? **The situation is worse in every sense that you could possibily imagine.** **The brazilian real value decreased 200%, and the pandemic out of control, and everything else that is no secret for the whole world.**

> Continue to use the PJZero. The PJZero has the same circuitry as the full HAT, except it has not HAT EEPROM, no dip switches and a different button configuration. **Ok! I will update it all and perform my tests :) As long as the wake up alarm, wake up on power works fine, for me is wonderful.**

Thank you Ton!!!