PiSupply/PiJuice #716: pijuice_cli cannot connect to PiJuice Zero

heri333, 2021-04-27:
I have a PiJuice Zero connected to a Pi Zero W, which seems to be working (powers the Pi, charges the battery). But `pijuice_cli` cannot connect to it. In the "General" menu entry it shows "Unable to connect to device", in other entries it just crashes, e.g. in "Status":
```
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2074, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 540, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1981, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 209, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 269, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 215, in get_status
    status = pijuice.status.GetStatus().get('data', {})
AttributeError: 'NoneType' object has no attribute 'status'
```
My current system information:
```
$ uname -a
Linux 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
```

```
$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 --
```
```
$ lsmod | grep i2c
i2c_bcm2708            16384  0
i2c_dev                16384  0
```

```
$ apt-cache show pijuice-base
Package: pijuice-base
Version: 1.7
```

tvoverbeek, 2021-04-28:
Somehow the pijuice does not get initialized properly in your pijuice_cli.
I noticed you have i2c_bcm2708 in your lsmod output.
For the current firmware you do not have to use the old i2c_bcm2708 module.
Try removing it from /boot/config.txt and see if the problem persists.
`lsmod | grep i2c` should then show `i2c_bcm2835`

heri333, 2021-04-28:
Yes, that was already an attempt to fix it. All other information remains the same when I revert that change:
```
$ lsmod | grep i2c
i2c_bcm2835            16384  0
i2c_dev                16384  0
```

tvoverbeek, 2021-04-29:
can you run this little python3 script in a terminal window:
```
#!/usrbin/python3

from pijuice import PiJuice

pj=PiJuice(1,0x14)
print(pj.status.GetStaus())
```
See if you get any error messages or if it produces the expected output.
Also run `i2cdetect -y 1` without sudo (should work without sudo if everything is installed correctly)

heri333, 2021-04-29:
Thank you very much for the hint regarding `i2cdetect`! I don't use the `pi` user, and after I've added the user to the groups `spi` and `i2c` the `pijuice_cli` works and your script prints
```
{'data': {'isFault': False, 'isButton': False, 'battery': 'NOT_PRESENT', 'powerInput': 'PRESENT', 'powerInput5vIo': 'NOT_PRESENT'}, 'error': 'NO_ERROR'}
```

closed 2021-04-29 by heri333

tig987, 2021-07-22:
Many thanks! Had the same issue and
the hint about groups `spi` and `i2c` saved my day!