PiSupply/PiJuice #720: Pijuice losing charge slowly (71% now)

arjanvlek, 2021-05-03:
Hi,

I've setup my PiJuice to use the correct battery profile (`BP7X_1820`) as the default profile was incorrect (1000 mah) and caused the battery to jump from like 70% to 20% when the power was removed.

Now, it charged only to around 80% at 4.05v, which I thought was to preserve the battery cells.
But since then, it has been slowly discharging and is now at 71% (3.935v). All of this while continiously being plugged in to the mains.

Is this correct behaviour or is something with my pijuice / battery going iffy?

See the attached screenshots of the current status and battery profile

[image]
[image]
[image]

arjanvlek, 2021-05-05:
UPDATE: It seems to have recharged to 80% (4.00v) now. Can anybody confirm if the 80% limit is set to make the battery last longer?

mmilann, 2021-05-05:
@arjanvlek It should charge approximately to regulation voltage, but you can see this only if battery is unloaded. Where you connected your power source? Source or adaptor may be weak, not able to provide enough current for powering Rpi and charging battery same time.

arjanvlek, 2021-05-05:
@mmilann I'm using the official raspberry pi 4 power supply and that is connected through usb-c. Besides the PiJuice, there is an Ethernet cable attached and one external 2.5 inch hard disk is attached to the USB port of my Pi. There are no monitor, keyboard and mouse connected as it is a server.

tvoverbeek, 2021-05-05:
@arjanvlek The load by the USB disk might be the culprit.
On my PiJuice + Pi4 (no peripherals) at charge level 95% shows 4.185V
Note the regulation voltage forthe BP7X-1820 is 4.18V
Also if the battery temperature goes over 45 deg, charge current is reduced. Over 50 deg charging is suspended.
The 45 deg is easily reached when using a Pi4B in a case.

arjanvlek, 2021-05-06:
@tvoverbeek Thanks for your comment.
The disk was not the culprit but the high temperature was.

Mine was constantly reporting 47 deg celsius. As a test I placed something cold (but not heavy) on top of the battery and as soon as it got below 45 deg (then reported 24 deg?!?), it started charging to 4.20v and 100% right away.

So if the temp is above 45 deg celsius, it will apparently not charge past ~80% (or 4.05v).
No idea though how to get this thing to run cooler. But such a high temp can never be good for the battery...

tvoverbeek, 2021-05-07:
@arjanvlek If this is a Pi4 use a small fan e.g. https://shop.pimoroni.com/products/fan-shim

arjanvlek, 2021-05-10:
@tvoverbeek Thanks for your suggestion and i'll look into that.

closed 2021-05-10 by arjanvlek