PiSupply/PiJuice #726: Bug in firmware 1.5?

rbuelund, 2021-05-09:
I just tried to update to firmware version 1.5. After I do that - the charge led is only blinking red a short time every few seconds. If I power through the juice, the raspberry restarts all the time.
I downgraded to v1.4 and now it works correct again. Something wrong with the 1.5 version ??

thesteg, 2021-05-10:
It sounds like it's in low power mode... when you say powered though the juice, does that mean previously it was powered though the RPi or on battery?

thesteg, 2021-05-10:
Ok, is the pijuice service running, and does it happen when you are using the piuice_cli utility?

rbuelund, 2021-05-10:
I follow this guide to update the firmware: https://github.com/PiSupply/PiJuice/blob/master/Firmware/README.md

rbuelund, 2021-05-10:
I can then go back to the V1.4 by following the FAQ at the bottom on the page, where i use: pijuiceboot 14 PiJuice-V1.4_2020_01_17.elf.binary

Then the status LED starts blinking blue/green again...

thesteg, 2021-05-10:
ok, if you rebooted then the service should become active again if you only issued the stop command to systemctl.

If the RPi is shutting down, that sounds like a setting somewhere to shut down the RPi, have a run through them with it powered via the pijuice, make sure power via gpio is enabled. The pijuice would go to low power mode once the RPi and flash the led every 4 seconds so sounds like the firmware is ok. 

If the led is red, chances are the battery is low.

rbuelund, 2021-05-10:
The battery is not low it is at 99% charge !??

thesteg, 2021-05-10:
ok, have a run through the settings. something is shutting the RPi down. Possibly even disable the system tasks to be sure.

rbuelund, 2021-05-10:
But why does it all work with firmware V1.4 and not with V1.5 ???

thesteg, 2021-05-10:
tbh, I'm not that familiar with the 1.4 firmware but it does seem that quite a few changes were made. It is possible that a shutdown feature only got included with 1.5. 

An OS shutdown of the RPi is instigated from the pijuice service running on the RPi not the firmware.

rbuelund, 2021-05-10:
Here are my settings:
[image]
[image]
[image]

tvoverbeek, 2021-05-10:
@rbuelund From your screenshots I see you are still on version 1.6 of the software.
For the new features of 1.5 you need to be on version 1.7, so please run `sudo apt update` and `sudo apt upgrade` to get the i.7 version of pijuice-base (and pijuice-gui if you are using the gui).
See also https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/changelog for all the cahnges in the firmware and software.