PiSupply/PiJuice #728: Pijuice gui not working  / starting up, low voltage warning

DieBe, 2021-05-14:
I bought a PiJuice Zero foer my Pi 4 and wired the necessairy pins (from here: [https://pinout.xyz/pinout/pijuice#](url)) up. I installed the gui as explained in your user guide using `sudo apt-get install pijuice-gui`
First problem: The Raspberry gives me nearly constantly a low voltage warning.
Second problem: The PijJuice status icon in the top bar isn't showing up, however, when I right-click on the space where it should be, I get, as expected, Settings and About...
Third problem: As I click on the Pijuice settings icon in the application menu, sometimes a little transparent window opens up for 1-2 seconds and closes itself again, the settings won't open up. Otherwise nothing happens. When trying to open the settings from the (invisible) icon in the top bar, I realised "Settings" was grayed-out, so I can't click on it.
I did try to reinstall the gui, but it didn't change anything. I'm using the 12000 mAh Battery.
Maybe I'm doing something completely wrong, can anyone help me?

tvoverbeek, 2021-05-14:
Which OS are you running on the Pi4? If it is a 64-bit one it is not going to work. Currently the software is 32-bit only.

DieBe, 2021-05-15:
I've just set the 32-bit version up to be sure. But it didn't change anything. Maybe the problem is that my Pi 4 has 8 gb of ram whereas 32-bit supports only 4 gb?

tvoverbeek, 2021-05-16:
The Pi 4B 8GB runs fine on a 32-bit OS.
Is on Raspberry Pi OS or Ubuntu or ...?

DieBe, 2021-05-17:
It is the newest Raspberry Pi OS.

tvoverbeek, 2021-05-17:
OK. Is the Pi4 running on battery (no external power)?
Which battery do you have connected to the PiJuice Zero?
The smaller PiJuice Zero batteries (500, 600, 1000, 1200 mAh) do not have sufficient capacity to power the more power hungry Pi4. If you use one of those you will most likely get the low voltage warning.

DieBe, 2021-05-17:
I'm using the big 12'000 mAh Batterie, as far as I have tested it, it doesn't matter wether the Pi4 is running from battery or not, but I'll check on that right now.

DieBe, 2021-05-20:
I've just got a Pi 3 A+, which I'll be using with my Pijuice. All the problems described above seem to be solved by switching to it. The only thing left is the low voltage warning, however it's got really rare.

ShonP40, 2023-06-26:
Getting a low voltage warning as well with my RPi 4 (4GB RAM) with the PiJuice Zero connected to a 10kmAh battery

DieBe, 2023-06-26:
I eventually managed to get rid of the warning by using shorter and thicker wires. Not sure how your setup looks, reducing resistance in your system in every possible way will help you.

ShonP40, 2023-06-26:
Didn’t think that these wires could cause issues but yea, I may try other ones [image]

ShonP40, 2023-07-03:
After checking it with a multimeter, it seems like this 10cm cable is dropping 0.2V (from 5V to 4.8V).
I'll try to find something better at some point probably.

DieBe, 2023-07-03:
From what I've experienced, the could be your main problem here. You may also want the check those, or if you can use something else.