PiSupply/PiJuice #731: "No module named pijuice"

FyHy, 2021-05-23:
I have raised this as a new issue as the previous issue on the subject (#204) was an issue between Python 2 and Python 3.
I am running Buster with Firmware 1.5 although this issue was present with 1.4. I have a Python 3 test script (courtesy of Ton) which is testing shut down and waking.  Pi Juice is imported but when being run other than from a command line, I get the error "No module named pijuice". It runs fine from a command line but not either from "Geany" or from the following crontab:
`@reboot sudo -u pi /home/pi/testscript.py 2>&1 | logger -t mycmd`

testscript.py: 

`#!/usr/bin/python3

from time import localtime, strftime, sleep
import pijuice
import os
import sys

while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

os.system("sudo hwclock -s")

pj = pijuice.PiJuice(1, 0x14)

chargelevel = pj.status.GetChargeLevel()['data']
stringtime_WU = strftime("%d-%m-%y %H:%M:%S", localtime())
with open('PJ_Actions.txt', 'a') as f:
    f.write('Woken up at:      ' + stringtime_WU + '\n')
    f.write('    Charge level: ' + str(chargelevel) + '\n')

sleep(30)

stringtime_Mid = strftime("%d-%m-%y %H:%M:%S", localtime())
with open('PJ_Actions.txt', 'a') as f:
    f.write('Mid Time at:      ' + stringtime_Mid+ '\n')

sleep(30)

stringtime_SD = strftime("%d-%m-%y %H:%M:%S", localtime())
with open('PJ_Actions.txt', 'a') as f:
    f.write('Shutting down at: ' + stringtime_SD + '\n\n')

# Ensure RTC wakeup is enabled in case it got reset (e.g. by timesyncd)
pj.rtcAlarm.SetWakeupEnabled(True)

os.system('sudo poweroff')`

I have checked the /dist-packages/ for pijuice.py and it is in "python3.7" but not in "python3".

Your advice please.

Geoffrey

tvoverbeek, 2021-05-23:
How did you install the pijuice-base package?
The normal install puts a symbolic link from /usr/lib/python3/dist-packages/pijuice.py to /usr/lib/python3.7/dist-packages/pijuice.py.
In your testscript there is a back-tick (`) before the '#!'. That should not be there.
With Geany you might have the opposite problem from #205. Geany starts a python script with `#!/usr/bin/env python` which will run python2.7 if you have not changed the default python to python3.
Your crontab entry is in the crontab for user pi or for user root?
If for user pi remove the `sudo -u pi`.

FyHy, 2021-05-23:
Thank you Ton.  I will go through those points and get back to you either
way.
FyHy

FyHy, 2021-05-24:
Hello again Ton,

The pijuice software would have been installed as in the Automated Process:

sudo apt-get install pijuice-gui

The "back tick" you refer to (there is one at the end of that piece of
coding) can only have been inserted as part of the insert coding process at
github which seems to have gone wrong.

The back tick is not there on the actual piece of coding in use on the Pi4.

I could obviously be wrong but my script run using Geany has
"#!/usr/bin/python3" does this not change the default?

Re: The crontab: I have tried the root user and the pi user with the same
result.  I have also tried using the crontab @reboot python(3) /home/pi/
etc as well as @reboot sudo python(3) /home/pi/ etc

The "(3)" is just to show that I have tried "python" and "python3" in both
situations - again without success.

I suspect that I will simply restart with a fresh Buster and a fresh
PiJuice and see where it goes. It used to work three or four months ago
so...

When advising me about previous issues you have suggested that
"permissions" may be  an issue. I accept that that is a huge topic but what
specific permissions should be where if the are critical?

Since you were last advising me on the shut down and wakeup problems that I
was having, I have introduced another piece of hardware (WittyPi3) into the
system and that is working fine but with certain restrictions over those
that the PiJuice should be able to provide me (e.g. UPS and I need more
than one i2c bus). So I have returned with  a new PiJuice to try again.

I will do a fresh install of Buster and Pijuice and let you know either way
if I succeed or otherwise.

Geoffrey.

tvoverbeek, 2021-05-24:
OK. Waiting for your results with the fresh install.
You are not running the 64-bit beta by any chance? PiJuice software is only tested on the standard 32-bit OS.
The install by apt takes care of setting permissions properly.

FyHy, 2021-05-24:
Thanks for that Ton.

I don't have an idea about any 64-bit beta so count that out.  That's
beyond my ability!!

I will do a fresh install this evening and get back to you.

Geoffrey

FyHy, 2021-05-26:
Okay...

I have done a fresh install of Buster and a fresh install of PiJuice using the "Automated" method set out in Software.
I below is a composite text file of the various commands followed by the results:

`pi@raspberrypi:~` $ sudo ps -ef | grep python
pijuice    446     1  0 22:09 ?        00:00:00 /usr/bin/python3 /usr/bin/pijuice_sys.py
pi         480   455  0 22:09 ?        00:00:00 /bin/sh -c python /home/pi/shutdown.py 2>&1 | logger mycmd
pi         491   480 98 22:09 ?        00:01:04 python /home/pi/shutdown.py
pijuice    769   762  2 22:09 ?        00:00:01 /usr/bin/python3 /usr/bin/pijuice_tray.py
pi         780     1  1 22:09 ?        00:00:00 /usr/bin/python3 /usr/share/system-config-printer/applet.py
pi        1309  1275  0 22:10 pts/0    00:00:00 grep --color=auto python

pi@raspberrypi:~ $ grep 'mycmd' /var/log/syslog
May 25 22:09:23 raspberrypi CRON[475]: (root) CMD (sudo python  /home/pi/testscript.py 2>&1 | logger -t mycmd)
May 25 22:09:23 raspberrypi pi: mycmd
May 25 22:09:23 raspberrypi mycmd: Traceback (most recent call last):
May 25 22:09:23 raspberrypi mycmd:   File "/home/pi/testscript.py", line 5, in <module>
May 25 22:09:23 raspberrypi mycmd:     import pijuice
May 25 22:09:23 raspberrypi mycmd: ImportError: No module named pijuice

pi@raspberrypi:/usr/lib/python3.7/dist-packages $ ls -l
total 76
-rw-r--r-- 1 root root   255 Feb 22 00:00 pijuice_base-1.7.egg-info
-rw-r--r-- 1 root root   249 Dec 24 00:00 pijuice_gui-1.7.egg-info
-rw-r--r-- 1 root root 61538 Feb 20 19:34 pijuice.py
drwxr-xr-x 2 root root  4096 May 24 22:42 __pycache__
pi@raspberrypi:/usr/lib/python3.7/dist-packages $ `

In /usr/lib/python3.7/dist-packages the following are present: pijuice_base-1.7.egg-info,  pijuice_gui-1.7.egg-info, and pijuice.py.
In /usr/lib/python3/dist-packages I see no sign of any Symlink from either to either. 
In /usr/lib/python3/dist-packages, there is only pijuice.py but not the other files.

Your advice please Ton.

Geoffrey

tvoverbeek, 2021-05-26:
Two things about your cron job:
`May 25 22:09:23 raspberrypi CRON[475]: (root) CMD (sudo python /home/pi/testscript.py 2>&1 | logger -t mycmd)`
- Why are you running the job as root? Use the crontab for user pi and do not use sudo in the command.
- Your current cron command uses python (=python2.7) and not python3. Hence the pijuice module is not found since it
only exists for python3.

The single pijuice.py file (link) in /usr/lib/python3/dist-packages is what is to be expected.

FyHy, 2021-05-26:
Thank you for that Ton. We have "lift-off" and the testscript.py routine is
working fine.

At some stage I will pull the power and see if it responds correctly to the
"Low Power" shutdown and then the "Wake-up" on charge with 1.5 which as you
may recall I was unable to achieve before.

I will let you know if that functions correctly.

Thanks for your help as ever.

Geoffrey.

FyHy, 2021-05-27:
Hello,
I know this is not concerning the current issue but I pulled the power and as before the text file shows that it just dropped way passed the low charge level and if not stopped manually would have depleted the battery completely.  Do you prefer that I mark this issue as solved and return to my previous issue to put forward a suggestion as to why the PiJuice may not be stopping at the low charge level?
Geoffrey

tvoverbeek, 2021-05-27:
@FyHy Yes, close this issue and raise a new one about the low charge level.
Please specify any relevant settings form the cli or gui and a listing of the used scripts and log files if possible.
The more info the better.

FyHy, 2021-05-27:
Okay - Thanks.

FyHy, 2021-05-27:
This issues has been solved by Ton and I am closing it.
FyHy