PiSupply/PiJuice #732: Schedule examples of Wake up functionality

tprinslo, 2021-05-24:
I find that the info on the reset of the wake-up alarm and when the flag will be reset as well as setting a daily schedule a bit confusing when I want to create a wake-up schedule not just a once off wake-up. 
Going through the forum answers reveals possible answers in different places for different firmware releases. 
I would like to have an example of what options to set in the GUI and CLI for the following scenarios:
Pi Wakes up every day at 6:00 
Pi Wake up every second day at 6:30
Pi Wake up Monday, Tuesday and Sunday at 16:00
What is the difference between the day and weekday option?
When will you tick the everyday tick box?
When will you tick every hour tick box?
What is valid options: When should a value be 0 and when should it be a space?

tvoverbeek, 2021-05-24:
**Note: All alarm times have to specified in UTC**

**- Pi Wakes up every day at 6:00**
  Tick the every day checkbox (Other day settings are ignored)
  Set hour to 6 (Every hour checkbox not selected) and minutes to 0 (Minute bullet selected)
  and seconds to 0
**- Pi Wake up every second day at 6:30**
   Not directly possible.
   You can select certain days of the current month e.g. '1;3;5;7;9;11' in the Day field (Day bullet selected)
   Since not all months have an even number of days 'every 2nd day' during the whole year does not work.
   But see next example.
**- Pi Wake up Monday, Tuesday and Sunday at 16:00**
  Select the Weekday bullet, Leave Every day checkbox unchecked
  In the day field specify 1;2;3. (1=Sunday, 2=Monday,...,7=Saturday)
  In the Hour field specify 16 (Every hour checkbox unchecked)
  In the Minute field (Minute bullet selected) specify 0
  In the Seconds field 0. 
**- What is the difference between the day and weekday option?**
   See examples above (day = day of month, weekday is what it says: Day(s) of week
**- When will you tick the everyday tick box?**
  With Every day checked it means every day. Day field and bullets are ignored.
  Hours, Minutes, Seconds as usual.
**- When will you tick every hour tick box?**
  Guess. Same logic as Every day.
  Note in the hour field (Every hour not selected) you can also specify a sequence e,g. 10:15:20
  as with weekdays.

tprinslo, 2021-05-24:
Thanks for the examples

closed 2021-05-24 by tprinslo

PacoJavi1, 2022-08-06:
I am looking at something similar. My goal is to have a solar powered raspicam which will take pictures everyday from 7 am to, say, 12 pm.

With another ups hat I got this set up running from minute 1, but I'm yet to find the specific configuration in the CLI to get this running with the pijuice.
Do I have to check the "Day" field and specify '7;8;9;10;11;12' in the "hour" field below?

Thanks

tvoverbeek, 2022-08-06:
@PacoJavi1 Use the every hour setting.At the end of the last (12pm) run reprogram the alarm to start at 7am.
At the end of the 7am run reprogram the alarm again for very hour.

PacoJavi1, 2022-08-06:
Thanks for the prompt reply. 
I am sorry but I am fairly new to this. What do you mean by "run reprogram the alarm"?

tvoverbeek, 2022-08-07:
The settings you can do in the GUI/CLI you can also do in your own Python script.
See the section https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-rtc-alarm in the documentation

gobees, 2023-05-03:
In my case, I want to wake up the pi every 20 minutes, Monday through Saturday between the hours of 0600 and 2000.  I believe based on the examples above, I would need to write code that checks RTC time to be at 2000 and when it is, my code reprograms the pijuice to wake up next at 0600.

Then I set up the day wake up schedule to include every Monday, Tuesday, Wednesday, Thursday, Friday and Saturday, and select in the minutes field 20 minutes 'period'.

Am I on the right track?

Thank you.

tvoverbeek, 2023-05-03:
Yes, that should work.