PiSupply/PiJuice #733: Question - Default Behaviour at 0% Charge

track02, 2021-05-27:
Hi,

I'm curious if there's any default behavior for the PiJuice (assuming service is running) if no Low Battery % event is specified and the battery reaches 0% with no external power provided

Will the unit shut the system down gracefully or is power simply cut if no Low % or event handler is specified?

Many thanks

mmilann, 2021-05-27:
Hi @track02 it will cut off power when battery voltage drops below minimum configured as battery profile parameter, ~3V for LiPo.

track02, 2021-05-27:
Thank you for very the quick response, as a related question - does the Min Charge event trigger whilst the battery is charging?

E.g.

- I setup Min Charge to trigger and shutdown at 10% battery
- The battery drains to below 10% and the Pi powers off
- If I resupply power and turn the Pi back on would it immediately power off again as the charge is still below the minimum amount?

mmilann, 2021-05-27:
It will not trigger shutdown if main power is present and battery is charging, only if absent and charge level is below threshold.

track02, 2021-05-27:
Excellent thank you, one final question (I promise!)

If I wanted to monitor for a power loss  and after a time period (30s) check whether power was still unavailable and make a decision. E.g. If power hasn't been restored shut the system down otherwise resume operation (power supply restored), I'm assuming that's something the PiJuice is easily capable of?

Going through the documentation I couldn't see anything quite matching this use case but I suspect this could simply be done via custom user script tied to the 'No Power' system event?

mmilann, 2021-05-27:
Potentially you can make user function (python script) and configure to trigger on NO_POWER event. That will setup system task with expiration period of 30 seconds, something like system service.

closed 2021-05-27 by track02