PiSupply/PiJuice #738: GetBAtteryCurrent

SebAcou, 2021-06-02:
Hello,
I am struggling to understand the unit of the current method : pijuice.status.GetBatteryCurrent()

When I am plugged , i've got -200 ( -20 in WEAK) and in [250, 300] while unplugged. The Io current is around 1.5 A.

It doesn't make any sense if it is mA, does it? 
My purpose is to assess different kind ok power supply -> solar panel,  charger.

B R,

jrhelbert, 2021-06-02:
The value is in mA, but I've found it to be a fairly unreliable number.

The PJ does not actually have the ability to measure battery current directly. There is no sense resistor to measure the voltage across (the IO Current passes through a ferrite bead that they have characterized for each board and they use that as a sense resistor.) The Battery Charger obviously knows what it's current charge current is, but it's I2C interface does not provide the ability to read the current, only the current limit.

Roughly, the FW in the PJ estimates the Battery Current based off of changes in SOC and the stated battery capacity. Ie, if you have a 1000mAh battery and the SOC starts at 90%, drops to 89.9% and then 1 min later it drops to 89.8%, then you know that in one minute you have used .1% * 1000mAh = 1 mAh. They can then use that times 1 min to extrapolate that the battery is discharging a 1mAh * 1min = 60 mA (60mAmin per mAhour.)

This starts to fall apart as you get to larger capacities and lower discharge rates since the reported SOC does not change very fast in these scenarios.

SebAcou, 2021-06-03:
Hello!
Thank you very much for your detailed answer.
The derivative of the SOC signal is a good idea. With a good Battery Charger and power through micro usb always "Present", we've got a nice sigmoid function ( 1820mHA). I guess it is a bit more tricky with a solar panel and power sometimes "WEAK".

[image]