PiSupply/PiJuice #742: RPI4 + Pijuice: Reboot under load

agjunyent, 2021-06-08:
Hi,

I've configured the Pijuice to run continuosly, and only act as emergency SAI, shuting down correclty the PI. Everything works perfectly with no load on the PI. I'm powering the PI's from their USB-C, as the current load is high and I've read PiJuice can only supply 2.5A MAX (I'm using some USB that has high power demands).

Problem comes when the PI is put under load. I use it as a ML "server" to process frames. I have 3 of them, each one with its own pijuice. Only one of them reboots after a while. It seems a "clean" reboot because there's no "dirty bit set" in dmesg after booting, so I think PiJuice is restarting it.

Does Pijuice have some kind of "overheating switch" to reboot the PI if temperature is too high? It only happens in one of them. Temperature reported by the PI using "vcgencmd measure_temp" is 75ºC MAX, when external temperature is high.

I cannot see any other problem. They run exactly the same code. PiJuice is configured exactly the same. In fact, they are all clones of the same SD card...

Thanks

tvoverbeek, 2021-06-08:
Your Pi4 is probably throttling under load.
Have you read https://www.raspberrypi.org/documentation/hardware/raspberrypi/frequency-management.md ?
You might need some cooling.

agjunyent, 2021-06-08:
Hi thanks for your response.

It's completely fine for me if it throtles down. But it is rebooting, not only throttling. And I'm completely sure it is rebooting because after it boots again automatically, looking at the syslog I see the complete reboot process (stopping services, lalalalal, starting services, kernel, boot).

If it was rebooting because of thermal protection, I think it simply shuts off, without correct reboot sequence, and the dirty bit is set (it is not set when this happens).

tvoverbeek, 2021-06-08:
Since you are powering via the Pi4 USB the only way for the PiJuice to initiate a shutdown is via the pijuice service running on the Pi. 
So there must be some event triggering the shutdown. What events do you have enabled and if so which one is connected to one ot the SYS_HALT_* functions or to a user function which initiates the shutdown. Also the system task needs to be enabled.