PiSupply/PiJuice #744: Pijuice immediately powers off on loss of power

devicenull, 2021-06-09:
I'm sort of at a loss here... I have a pijuice connected to a raspberry PI 4, and as soon as I disconnect power the pijuice appears to power off the Pi, without executing the user script.

I've even stopped the `pijuice.service` task and verified nothing with the name pijuice is still running.

What else could be causing this instant power off?  It certainly appears to me that it's graceful - the activity lights on the Pi appear to continue flashing for ~10s .

I'm running firmware v1.5_2021_02_06, and using Raspbian as the operating system.

As far as the Pi itself goes - I don't have anything plugged into the usb ports except and externally powered USB hub.  This USB hub also powers the Pi itself.  I suppose it's possible the hub is failing over to the USB port for power - however would that cause an instant graceful poweroff if the power draw exceeded some limit?

I'm open to whatever ideas - I've definitely run out of things that I can troubleshoot here.

devicenull, 2021-06-10:
FWIW, I added some debugging to pijuice_sys.py, it seems like it is seeing this up until it powers off:

> {'data': {'isFault': True, 'isButton': False, 'battery': 'CHARGING_FROM_IN', 'powerInput': 'PRESENT', 'powerInput5vIo': 'NOT_PRESENT'}, 'error': 'NO_ERROR'}

I've also tried with the Pijuice connected to a different power source then the USB hub - if I just disconnect the Pijuice, it still powers off instantly, which makes me believe this is not an overcurrent issue.

devicenull, 2021-06-10:
I just downgraded to firmware v1.4, and it appears to be acting the same way.

[image]

tvoverbeek, 2021-06-10:
From your first screenshot:
You are obviously powering the Pi from the PiJuice and not from the USB hub (GPIO power not present)
The current from the PiJuice to the Pi4 is very high: 3.1A.
A Pi4 with nothing connected should only use something less than 1A.
I suspect something fishy with your power setup.

devicenull, 2021-06-12:
I assumed that was related to the battery still charging.  However, even with nothing plugged in but a single flash drive and charging disabled, I'm still seeing anywhere from 1-2A of current draw from GPIO.

However, I have a [url=https://www.amazon.com/gp/product/B07JYVPLLJ/]USB power anaylzer[/url] that isn't really matching the numbers provided by the PiJuice - it's showing me a 0.6-0.7A draw, while the Pijuice is showing about 2x that.  I unfortunately don't have a third device to check and see which is correct!

Would overcurrent even explain the immediate shutoff without executing the script?  It's not as if the Pi itself immediately dies when power gets lost - based on the LEDs it seems to go through some sort of disk sync and shutdown.  I have not yet attached a monitor to see what's happening.