PiSupply/PiJuice #750: Max power never reached

lesthawk, 2021-06-17:
Since powering up the battery has only once hit 100%. It seems unwilling to charge past 79%, ui says power is present both on gpio and on usb.
What can i do to make sure the battery gets fully charged until it is actually needed?

tvoverbeek, 2021-06-17:
Have you checked in the UI that charging is enabled?

lesthawk, 2021-06-18:
Which tab and which setting? The radio button is set for usb but I’ve tried it on gpio as well.

tvoverbeek, 2021-06-18:
Same tab, last line: Charging enabled + check box

lesthawk, 2021-06-18:
> Same tab, last line: Charging enabled + check box

Yes. It’s checked.

tvoverbeek, 2021-06-18:
OK. Which battery? Which PiJuice (HAT or Zero)?
Can you make a dump of the status page (this one; https://github.com/PiSupply/PiJuice/tree/master/Software#main-software-menu) or write down the contents?

lesthawk, 2021-06-18:
Default battery that came with it I believe to be 1800 ma
Normal status, present for both gpio and usb, voltage consistent and 5v on usb
Values can change to a minimal amount ie decimal places.

4.022V (varies up to 4.038) 51 degrees Normal 
Power input: 5.0v, -0.1A to -0.3A, PRESENT USB Micro power input: PRESENT
Fault: no fault

lesthawk, 2021-06-18:
My concern is the firmware doesn’t seem to be switching to charge it to 100% what settings in the hat config would possibly limit this?

tvoverbeek, 2021-06-18:
So you are seem to be using the HAT with the default BP7X 1820 mAh battery.
Your problem is the battery temperature: 51 degrees.
If you look at the battery tab with the battery parameters you see that for he BP7X
the warm temperature is 45 and the hot temperature 59.
Above the warm temperature charge current is reduced and above hot suspended.
Your battery status is reported as NORMAL, ie it is not charging.
If it is charging it should say charging_from_in (=PiJuice usb) or charging_from_5v_io (=from the GPIO).
Also why do you have 2 power connections (one to the RPi and one to the PiJuce)?
Try to cool down the battery and use only one power connection

lesthawk, 2021-06-18:
2 power connections reasons
if I disconnected the mains to the pi and only charged from the battery the battery discharged slowly and dint charge up to max
If I disconnected the power to the battery the battery didn’t seem to charge from the gpio at all.
This has been like this a week. Just stuck at 79% now and not moving.

lesthawk, 2021-06-19:
Also worth mentioning, power by gpio AND usb seems supported in the hat config, why else would there be an ‘input precedence’ option?

tvoverbeek, 2021-06-19:
Yes, you can have both inputs connected.
But your main problem is the measured temperature of the battery which limits/inhibits the charging.

Is the RPi/PiJuice in a closed case? Especially if it is a Pi4 doing some heavy lifting it can get hot.
What you can try (if you are sure the battery is not too hot in reality) is to set the 'Temperature Sense' on the
Battery tab to NOT_USED. Then the firmware will use a default battery temperature of 25 degrees
and you should get a charging indication on the general status page (... CHARGING_FROM_...)
Also which software version (= package version) and firmware version are you using?
There are battery temperature measurement fixes in the latest firmware/package version.

lesthawk, 2021-06-19:
Firmware fully up to date, haven’t got a lot on the pi but it does seem to get quite warm none the less. 

Complete setup explained:
Running inside an open soho cabinet under stairs with a 3D printed 1u custom shelf holder for the pi4 4gb, but does have a full set of USB ports in use. All because it’s running samba-addc, pihole for a number of devices and a printer server, however load is low overall whilst.

lesthawk, 2021-06-19:
Using your advice I turned off the temperature sensor and got the charge to 96%, periodically checking the sensor on found the temp stayed at 50 degrees.
However I can’t seem to get it to 100% and it’s stuck at 96% 4.157v, NORMAL (so not charging).
Gpio currently at 5.034 - 5.039 volts, -2.2 to 0.8 amps
Both usb and gpio capabale of power.
No fault showing.

lesthawk, 2021-06-19:
Firmware version 1.5 the firmware file path indicates from the binary name that it appears to be from 06/02/2021

lesthawk, 2021-06-27:
> Yes, you can have both inputs connected.
> But your main problem is the measured temperature of the battery which limits/inhibits the charging.
>
> Is the RPi/PiJuice in a closed case? Especially if it [...]

Have managed to get battery to 95% by disabling the sensor, however once back on even with the temp showing at 50 afterwards it's discharging and not getting back to the expected levels, currently at 74%