PiSupply/PiJuice #751: No Power Event - Not firing when booting from battery only?

track02, 2021-06-23:
Hello,

I've been testing the `No Power` event and mapping it to `SYS_FUNC_HALT_POW_OFF` so the board powers down once mains supply is lost and this looks to work great

However I've noticed that if no mains power is available and I turn on the Pi via Pi Juice's SW1 the Pi will boot up and stay powered on using the battery until its charge is depleted

Is there a recommended way to prevent the Pi from booting if no power is detected? Or if that's not possible, ensuring the `No Power` event triggers and shuts the Pi down shortly after it's booted?

We're mainly interested in using the Pi Juice to allow for clean shutdowns rather than for a battery powered device

Many thanks

tvoverbeek, 2021-06-23:
Yes, the current release version (1.4) will no trigger 'no power' when booting from battery.
This was also covered in discussion #701.
Current github version of pijuice_sys.py will trigger no power when booted from battery.
Note especially the comments on lines 39 and 40 of https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py.
Note also the updated github versions of pijuice_cli.py and pijuice_gui.py to enable the 'power present' event.
However if you only need the no power event to trigger when booting from battery you only need the updated pijuice_sys.py.

track02, 2021-06-23:
Thanks for the quick response, I'll swap out `pijuice_sys.py` tomorrow for the latest version and see if that sorts it

Also, is there an estimate for when the next release (1.5?) will be that includes this change as default?

track02, 2021-06-24:
I've swapped `pijuice_sys.py` out of `/usr/bin` for the current Github version, following the discussion in the linked issue (701) to change permissions, however the service no longer seems to start correctly `systemctl status pijuice.service` gives the following:

```
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2021-06-24 08:54:30 BST; 39s ago
  Process: 835 ExecStart=/usr/bin/pijuice_sys.py (code=exited, status=203/EXEC)
  Process: 836 ExecStopPost=/usr/bin/pijuice_sys.py stop (code=exited, status=203/EXEC)
 Main PID: 835 (code=exited, status=203/EXEC)

Jun 24 08:54:30 gateway systemd[1]: pijuice.service: Service RestartSec=100ms expired, scheduling restart.
Jun 24 08:54:30 gateway systemd[1]: pijuice.service: Scheduled restart job, restart counter is at 5.
Jun 24 08:54:30 gateway systemd[1]: Stopped PiJuice status service.
Jun 24 08:54:30 gateway systemd[1]: pijuice.service: Start request repeated too quickly.
Jun 24 08:54:30 gateway systemd[1]: pijuice.service: Failed with result 'exit-code'.
Jun 24 08:54:30 gateway systemd[1]: Failed to start PiJuice status service.

tvoverbeek, 2021-06-24:
Not really. For me it works fine.
Can you try to run it manually in a terminal window:
`python3 /usr/bin/pijuice_sys.py` and see if you get an error message which could help.

track02, 2021-06-24:
So running manually outputs the following and nothing else, I assume it's just running:

```
sudo python3 /usr/bin/pijuice_sys.py 
rtc_ds1307 mdule reloaded and RTC os-support OK
```

And my /usr/bin/ looks like this:

```
...
-rwsr-xr-x 1 pijuice pijuice     9096 Dec 17  2018  pijuice_cli
-rw-r--r-- 1 pijuice pijuice    98358 Feb 11  2020  pijuice_cli.py
-rw-r--r-- 1 pijuice pijuice    15664 Jun 24 08:51  pijuice_sys.py
...
```

track02, 2021-06-24:
Had another look and may have fixed it, I reinstalled  `pijuice-base` and checked `/usr/bin` noticing the permissions were slightly different 

```
-rwsr-xr-x 1 pijuice pijuice     9096 Dec 17  2018  pijuice_cli
-rw-r--r-- 1 pijuice pijuice    98358 Feb 11  2020  pijuice_cli.py
-rwxr-xr-x 1 root    **root**       12051 Feb 11  2020  pijuice_sys.py
```

So copying the GitHub `pijuice_sys.py`, adding write permission and leaving owner as root looks to be the way to do it, will do some further testing to confirm

tvoverbeek, 2021-06-24:
The missing permission is the execute permission (x).
`sudo chmod 755 /usr/bin/pijuice_sys.py`
That should do it.

track02, 2021-06-28:
Yes that's all it was, tested it a few times since and had no problems, I'll close this issue

Thanks for the help

closed 2021-06-28 by track02