PiSupply/PiJuice #752: invalid temperature reading

rsie-dev, 2021-06-26:
My setup:
- pjuice zero W
- pijuice zero 1000mAh battery
- firmware 1.5
- raspbian buster (10.9)

All settings at default.
The battery won't charge as the temperature read is constantly below zero (e.g. -8°C)
The actual ambient temperature is +25°C

tvoverbeek, 2021-06-26:
What battery profile is selected?
Which temperature measurement selection (on the battery tab)?
Which software (package) version?
Did you try to change the temperature measurement selection?
If you select 'NOT-USED' the firmware assumes 25 deg C and the battery should charge.

rsie-dev, 2021-06-27:
> What battery profile is selected?
the profile is PJZERO_1000

> Which temperature measurement selection (on the battery tab)?
The selected sense is AUTO_DETECT
But I get the same readings when I switch to NTC.

> Which software (package) version?
pijuice-base: 1.7

> Did you try to change the temperature measurement selection?
> If you select 'NOT-USED' the firmware assumes 25 deg C and the battery should charge.
Yes, that is correct and does activate battery charging.

With a strong power source I've noticed however that the battery gets quite hot  (to the touch).
The planned installation is not constantly monitored so I would like to have some safety checks when charging.

tvoverbeek, 2021-07-14:
Smells like a problem with the battery. NTC not connected or not working would give you a very low temperature when using the NTC measurement. Also the battery getting hot is not right.