PiSupply/PiJuice #756: pi3b not connecting to hat

cumafo, 2021-07-05:
Running a fresh installed pi3b+ 
uname -a: 
Linux raspberrypi 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux

apt update && apt install apt install pijuice-base
dpkg -l|grep juice
ii  pijuice-base      1.7    all    Basic support for Pi-Supply's PiJuice HAT

Tried factory reset and suggestions as in issue #716 but no luck.
The hat works booting and halting using sw1.

```
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

```
lsmod | grep i2c
regmap_i2c             16384  1 rtc_ds1307
i2c_bcm2835            16384  1
i2c_dev                20480  2
```

```
#!/usr/bin/python3
from pijuice import PiJuice
pj=PiJuice(1,0x14)
print(pj.status.GetStaus())
```

Traceback (most recent call last):
  File "./chk-juice.py", line 6, in <module>
    print(pj.status.GetStaus())
AttributeError: 'PiJuiceStatus' object has no attribute 'GetStaus'

ChristopherRush, 2021-07-05:
There is a typo in your command getstatus

cumafo, 2021-07-05:
Not sure what you mean. That's a copy/paste from #716 (which itself has a typo, missing slash)
Anyway, nothing in pijuice_cli works

Status gives:
```
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2074, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 540, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1981, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 209, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 269, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 215, in get_status
    status = pijuice.status.GetStatus().get('data', {})
AttributeError: 'NoneType' object has no attribute 'status'
```

ChristopherRush, 2021-07-05:
In your Python script chk-juice.py there is a typo GetStaus should be
GetStatus.

cumafo, 2021-07-05:
OK - thanks, that gives:
{'data': {'isFault': False, 'isButton': False, 'battery': 'NORMAL', 'powerInput': 'PRESENT', 'powerInput5vIo': 'BAD'}, 'error': 'NO_ERROR'}

ChristopherRush, 2021-07-05:
Yes, that it the expected response from command 'GetStatus'.

cumafo, 2021-07-05:
So any ideas of why all menu-cmds in pijuice_cli  fails?

tvoverbeek, 2021-07-05:
Are you running as user pi or as an other user?
What happens if you run the following in a terminal window;
`sudo -u pijuice python3 /usr/bin/pijuice_cli.py`?
Any error messages will be mixed into te pijuice_cli output.

cumafo, 2021-07-05:
OK - running as root was the issue.
Thank you!

closed 2021-07-05 by cumafo