PiSupply/PiJuice #757: Early access to rtc

cumafo, 2021-07-06:
According to the guides I have found, the rtc is read from pijuice and set at rc.local which is too late for my application.
Is it possible to get access to it earlier during the boot/systemd process?
pi3b+ , 5.10.17-v7+

tvoverbeek, 2021-07-06:
What do you want? Setting the RTC from the system clock or setting the system clock from the RTC?

Note that the (simulated) RTC in the PiJuice is not available until I2C is initialized.
Usually /etc/rc.local is started before I2C is available.

cumafo, 2021-07-06:
Set system clock from rtc. Any way then to initialize i2c earlier?

tvoverbeek, 2021-07-06:
If you want to do it as soon as possible see https://github.com/PiSupply/PaPiRus/blob/master/RTC-Hat-Examples/README.md
which also uses an I2C based RTC. Using udev the system clock is set from the hardware clock as soon as the kernel
recognizes the device.
This requires using either EEPROM address 0x50 so the kernel module is loaded (only for the HAT) or use an explicit
`dtoverlay=i2c-rtc,ds1339` in /boot/config.txt (required for the PiJuice Zero).
Of course the fakehwclock package has to be removed.

cumafo, 2021-07-12:
Thanks a lot.

closed 2021-07-12 by cumafo