PiSupply/PiJuice #758: Measure current and voltage coming from micro USB

EilertHe, 2021-07-08:
Hello, 

I am working on a project with the Pijuice connected to a solar panel through the micro USB. 
One of the features that I would like is the capability to measure the power output from the solar panels(the power coming into the micro USB port). Is this possible in some way with the Pijuice? 

I am aware of the .GetIoCurrent and .GetBatteryCurrent python commands, but I dont completely understand how the circuitry work. For example if there is sun outside and the solar panel is delivering power to the Pijuice(whilst RPI is running), will the power from the micro USB go to the battery or will it be used by the PI directly? In other words does the battery charge and discharge at the same time? 

Thanks for all input on this topic :)

tvoverbeek, 2021-07-08:
There is no direct measurement of the power at the micro-USB (or J5) input.
The power fom the input goes to the charging control IC (BQ24160RGET) which directs (part of) the input to the battery for charging. The remainder (and/or battery output if not charging) goes to the system (VSYS).
A DC-DC converter is used to provide the 5V to the GPIO for the Pi.
Alternatively if power comes from the Pi it goes also to the charge control IC.
For the nitty gritty see https://github.com/PiSupply/PiJuice/blob/master/Documentation/PiJuice_Schematic.PDF.

Note: the GetBatteryCurrent and GetIoCurrent outputs are based on estimates. There is no direct measurement.

EilertHe, 2021-07-08:
Thanks, Ton! 
But if the battery can charge and discharge at the same time, will the GetBatteryCurrent more often than not show charging estimate and not discharge estimate?

Could one assume that if the battery is charging(current going into battery) that the IOcurrent is remaining current from PV panels and if the panels dont supply enough power the battery will deliver the rest ?

I.e. the napkin math could be something like this:

P_solar=P_batteryCharge + P_IO - P_batteryDischarge + (P_losses)

Sorry if this seems messy or if Im completely off, just trying to understand how the PIJ works :)

tvoverbeek, 2021-07-08:
Battery cannot charge and discharge at the same time.
I would use the following napkin math:

P_solar=P_battery+P_IO+P_losses

P_battery positive when charging, negative when discharging.
E.g. during the night (P_solar = 0) P_battery should be -(P_IO+P_loses) i.e. discharging
P_losses includes the power needed to run the PiJuice circuitry and the losses of the voltage converters.

EilertHe, 2021-07-08:
Thanks again, that was what I was looking for 😄

EilertHe, 2021-07-30:
Hi, Ton(or anyone else with experience with the PiJuice) 

I understand that the Get commands are just estimates and should not be trusted to much, but I ran a test were the PI4 was only supplied with power from the standard battery. I then plotted the voltage and current for both battery and IO. Shouldnt the power output from the battery be higher than whats going thorugh the IO pins? 
I am not that familiar with integrated circuitry. 

picture for reference: 
Voltage[mV], current[mA] and charge[%] plotted with their respecting y-axis over time. 

[image]