PiSupply/PiJuice #759: Need a bit of help understanding charge current

Vauxha, 2021-07-17:
I have fitted my PiJuice with the standard battery to my pi model 3 with Approved 2.5 amp power supply powering the pi directly.
I did a quick check to test things out. Switched power from mains off and battery took over showing current draw between 0.5 and 1.0 amps. Let the the battery run down to 50% and powered mains back on. GPIO power input shows current draw between -1.5 and 3.5 amps WEAK. I don’t understand why the figures are so high.
Thanks

tvoverbeek, 2021-07-17:
What are the GPIO Power Input Voltage and Current when you are charging and you get the WEAK indication?

Vauxha, 2021-07-17:
4.961V -2.9A

tvoverbeek, 2021-07-17:
Something is not right. 2.9A going to the PiJuice with BP7X from the RPi is far too much.
What happens if you connect the power supply directly to the PiJuice?
Does it then charge/discharge properly?

Vauxha, 2021-07-18:
I have connected power to the PiJuice as suggested and carried out the same test  as my previous comment. With the battery at 50% now shows a current varying between 0.5 and 1.5 amps and no indications of weak power. The battery won’t now charge past 98% and shows continual charge.