PiSupply/PiJuice #760: Wake on charge 'restore' option does not work on battery drain/battery removal

olivermonberg, 2021-07-19:
Hi, 

I'm having an issue when trying to wake my RPI 4 on charge, when I simulate a battery drain scenario. I was hoping that the 'restore' option would fix my case, but it seem my RPI could still be left unbooted.

I have set the battery profile buttons on the board and the battery profile in the GUI to fit my battery (BP7X).
Running with firmware version 1.5.

What I did:

1. Set system event 'Low charge' to SYS_FUNC_HALT_POW_OFF
2. Set minimum charge to 10%
3. Set wake on charge to 15% with the 'restore' option enabled.
4. My RPI is running with a charge above 10%
5. I unplug the power from the micro USB on the PiJuice
6. My RPI shutdown when charge is below 10%
7. I wait 60 seconds for the power to be completely cut to the RPI
8. I remove the battery (battery drain)
9. I reconnect power to the micro USB on the PiJuice and immediately reconnect the battery.
10. I can see on the LED's on the PiJuice that the battery is recharging. 
11. I wait for the RPI to boot on 15% charge, but this does not happen.
12. My RPI is left unbooted when though power is present and battery is charged.

I was not expecting this after reading various threads here on GitHub, talking about storing wake on charge configuration in non-volatile memory. Threads such as #357 #612 #680 

Can any of you shed some light on this behavior? Thanks.

mmilann, 2021-07-19:
Good catch @olivermonberg . If it is powered from input source before battery it will miss to enable wakeup if "No battery turn on" option is disabled.

olivermonberg, 2021-07-19:
Thanks for a quick reply @mmilann.
All right, but would this not be the case if the battery is completely drained and still connected to the PiJuice? 
When reconnecting power, would it not still miss to enable wakeup if "No battery turn on" option is disabled?

mmilann, 2021-07-19:
Yes it will be case, because it evaluates end enable wakeup config during period where there is no power input present or immediately after power-up if power input is not present. For now if it works for you enable "No battery turn on" option so it will turn on Rpi immediately after PiJuice power-up from power source and no battery is present.

olivermonberg, 2021-07-19:
Okay, thanks. 
So with "No battery turn on" option enabled and in the event of a unexpected power cut immediately after power up, it could result in the RPI not being safely shutdown?

olivermonberg, 2021-07-19:
Okay, well it could still happen. If the PiJuice waited to turn on the RPI until the the battery recharged to the wake on charge value, it would never end up being switched off unsafely. 

Pardon me if I misunderstood some things about the "wake on charge" restore option. But what I have understood was that restoring the wake on charge value from non-volatile memory would enable the PiJuice to boot the RPI on the wake on charge value after a complete battery drain scenario?

mmilann, 2021-07-19:
Yes that would be issue and is for consideration on next firmware update, based on observations like your one. For now it works in conjunction with "No battery turn on" enabled. Usually PiJuice cuts 5V power and goes to standby if battery voltage becomes to low in order to prevent total drain but still may happen if too long without power source.

olivermonberg, 2021-07-19:
Okay, thank you.
So just to understand you correctly, the "wake on charge" restore option does not have any effect at the moment with the current firmware?

mmilann, 2021-07-19:
It has effect, you do not need to use software command each time system boots to enable it, because it is from V1.5 automatically enabled after power source disappears, if restore is checked, so it is more reliable.

olivermonberg, 2021-07-19:
Of course, thanks a lot, I am a lot wiser now. I still have three questions though, sorry.

1. Is it possible to set the wake on charge restore option in the JSON config file?

2. I am reading incorrect charge percentage in the GUI. I have "No battery turn on" enabled and then when I connect power (RPI boots) and then plug in the battery, even though the battery is nearly drained, the charge in the GUI reads high percentage (e.g. 80% charged). The battery profile is correct both in the GUI and the buttons on the back of the HAT. A reboot of the RPI does not make a difference. If i shutdown the RPI, unplug battery and power, reconnect battery then reconnect power, RPI boots and everything is fine and normal.

3. I am getting testing software insted of stable when i run sudo apt-get install pijuice-gui. Could the incorrect charge reading be related to this?
[image]

mmilann, 2021-07-20:
1. No. You need to use gui or cli and set restore only once. 
2. This is expected, charge estimation algorithm does not know initial state, but it will converge to correct value. 
3. Charge reading should not influence software version.

olivermonberg, 2021-07-21:
Okay, thanks! But can you confirm that the PiJuice software that I have installed is not the stable 1.7 release?

mmilann, 2021-07-21:
It's ok if you installed from official Rpi repo. If you rebuilt from github sources, there could be updates made for testing.