PiSupply/PiJuice #765: low power showing on pi4 when using pi juice even if powered with usb

rallycorsa, 2021-07-23:
[image]

this is a screen shot of my pi4 hooked up to pijuice with genuine pi supply powering the pi its showing weak and the current is negative the pi keeps shutting down then as I have code set for when supply is missing use UPS to shut down pi after 60 seconds

any way to fix this

tvoverbeek, 2021-07-23:
What peripherals have you connected? 
External disks (also SSDs) use quite a bit of power when booting.

rallycorsa, 2021-07-23:
Yes its hooked up us a usb3 solid state drive

rallycorsa, 2021-07-24:
I ran the python code, with SSD drive, and running off a genuine PI4 mains plug, showing fault, in the API pijuice it showing 5.16V -0.9A and weak, any suggestions ideally I want to run my project off the micro usb on the pijuice if anybody can recomment a supply suitable to go into a car application 

Status =  {'isFault': True, 'isButton': False, 'battery': 'NORMAL', 'powerInput': 'NOT_PRESENT', 'powerInput5vIo': 'WEAK'}
Fault =  {'forced_power_off': True}
Charge = 99 %, T = 32 , Vbat = 4170 , Ibat = 24 , Vio = 5148 , Iio = -776
Battery temperature sense = AUTO_DETECT

tvoverbeek, 2021-07-24:
Suggest to use a higher capacity automotive 12v->5V converter.
Something like https://www.amazon.com/dp/B08MCCSP38
which has a max output of 10A on 5V