PiSupply/PiJuice #766: NTC practically takes ON_Board temperature readings

ionkarolos, 2021-07-29:
Hello,

I am using a regular pijuice hat with a raspberry pi 4. The firmware version in my pijuice is the latest, 1.4. I have also attached to the pijuice an official 5000 mah battery. In the pijuice configuration menu, I have also checked the battery profile for the 5000 and NTC temperature readings. My setup is inside a fan ventilated 3D printed case. My battery temperature readings are very high, about 61 Celsius degrees. The problem is that when I change the NTC to on board setting for temperature reading the temperature stays the same. This is unusual high readings and I think that they come from the pijuice PCB itself and not from NTC, because of the closed distance between pijuice and Broadcom chipset of rpi4. When I reset everything on EEPROM of pijuice and I selected again NTC, the temperature goes immediately to 42 degrees of Celsius which is normal. But again after a few minutes, the temperature goes from 42 to 61! The battery itself is cold in touch and I don’t know exactly what’s going on. Can you help me? Thanking you in advance.

tvoverbeek, 2021-07-29:
Which version of the software and firmware are you using?
In the current versions (software 1.7, firmware 1.5) there are some fixes to the battery temperature reporting.

ionkarolos, 2021-07-29:
[image]
I am using the (software 1.7, firmware 1.5). Please find attached the related picture.