PiSupply/PiJuice #767: pijuice-cli 1.7 not working on raspberry pi zero w

GerBreOwn, 2021-07-30:
I just installed the pijuice hat on my raspberry pi zero w along with the pijuice-base v1.7 program. 
I su pi to run the program. 
I get a menu screen but when I select any option I get a lot of error messages and the program crashes.

Do you have a newer version?

Thanks.

Gerald S. Brown

tvoverbeek, 2021-07-30:
No newer version.
What version of Pi OS (Buster or ???)
What do you mean by 'I su pi'?
And do you really use the HAT or the PiJuice Zero (= same size as Pi ZeroW)?
If possible catch the error messages, that could give me a cue on what is going on.

bb-ate, 2021-08-01:
I'm also having a similar problem with the base/cli program on a Pi Zero with the full size PiJuice HAT. I'm running DietPi (based on Raspbian Buster). I'm logged in as root over SSH. If I go to any menu item except 'General' I get a python exception along the lines:
```
  File "/usr/bin/pijuice_cli.py", line 2074, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 540, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1981, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 209, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 269, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 215, in get_status
    status = pijuice.status.GetStatus().get('data', {})
AttributeError: 'NoneType' object has no attribute 'status'
```
If I go to 'General' I get 'Unable to connect to device'.

But if I do a 10-sec press of SW1 I am able to initiate a controlled shutdown of the pi, so I'm assuming the device is connected and communicating OK?

bb-ate, 2021-08-01:
Only after posting that have I notice the bit in the documentation about needing to run as user 'pi'! But as DietPi comes with dietpi rather than pi I'm not sure where that leaves me. Creating the 'pi' user after installation hasn't worked as I suspected.

GerBreOwn, 2021-08-01:
> No newer version.
> What version of Pi OS (Buster or ???)
> What do you mean by 'I su pi'?
> And do you really use the HAT or the PiJuice Zero (= same size as Pi ZeroW)?
> If possible catch the error messages, that could give me a cue on what is going on.

GerBreOwn, 2021-08-01:
> No newer version.
> What version of Pi OS (Buster or ???) Buster Lite
> What do you mean by 'I su pi'? It says that the program has to be run as the pi user so I "switch user" (su) to pi as I log in as another user.
> And do you really use the HAT or the PiJuice Zero (= same size as Pi ZeroW)? It is the HAT as it is twice as big as the Pi Zero. Didn't know there is the pijuice zero available.
> If possible catch the error messages, that could give me a cue on what is going on. See the attached file "errors.txt". It is a full screen full of errors when I click on any option in the pijuice_cli menu.
Thanks
[errors.txt](https://github.com/PiSupply/PiJuice/files/6912491/errors.txt)

tvoverbeek, 2021-08-01:
@GerBreOwn @bb-ate Both error traces are caused by an uncaught Permissions error on accessing the i2c bus.

If you used the 'apt install' the post-install script should have created a new user pijuice and added it to the i2c group.
Please check if user pijuice exists and belongs to the i2c group.
The configuration file is owned by user pijuice and is not readable by other users.
pijuice_cli is a setuid executable owned by pijuice and runs python3 /usr/bin/pijuice_cli.py as user pijuice.
Check the output of `ls -l /usr/bin/pijuice_cli*`.
Expected output:
```
pi@raspberrypi:~ $ ls -l /usr/bin/pijuice_cli*
-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_cli
-rw-r--r-- 1 pijuice pijuice 101267 Aug  1 21:19 /usr/bin/pijuice_cli.py
pi@raspberrypi:~ $ 
```
Also your default user should belong to the i2c group.

GerBreOwn, 2021-08-02:
I followed the above instructions. All looks as specified.

I was about to say good news as the first 3 options worked!!! However when I went to option 4 or greater I started getting a lot of error messages again.

There is some progress but still no go.

Do you need to see the error messages again?

tvoverbeek, 2021-08-02:
@GerBreOwn What type of error messages? COMMUNICATION_ERROR?

bb-ate, 2021-08-02:
For my problems with DietPi OS, adding the standard user to the i2c group  as you suggested, worked:
`sudo usermod -a -G i2c dietpi`

Thank you.

GerBreOwn, 2021-08-03:
It does NOT seem to be a communication error.
Attached is the error file.
[error2.txt](https://github.com/PiSupply/PiJuice/files/6923303/error2.txt)

tvoverbeek, 2021-08-03:
Could still be a COMMUNCIATION_ERROR if the statement on line 647 of pijuice_cli.py:
`result = pijuice.config.GetLedConfiguration(self.LED_NAMES[i])`
returns with result containing an error message and no 'data' field.
On which RPi are you running? Pi4?

GerBreOwn, 2021-08-03:
RPi Zero WH

GerBreOwn, 2021-08-03:
I am SSHing from my laptop to the RPi Zero if that makes any difference as the pijuice_cli.py is running on the Zero not the laptop.

tvoverbeek, 2021-08-03:
Just to check if the GetLedConfig works, can you try to run the following python script in a terminal window:
```
#!/usr//bin/python3

from pijuice import PiJuice
pj=PiJuice(1,0x14)
print(pj.config.GetLedConfiguration('D1'))
print(pj.config.GetLedConfiguration('D2'))
```
and see if you get sensible output

GerBreOwn, 2021-08-03:
That code returns:
{'error': 'COMMUNICATION_ERROR'}
{'error': 'COMMUNICATION_ERROR'}

The only thing I have connected to the pijuice is a larger battery. I do have some wires connected to the pins on the juice but the other ends are not connected to any thing at the moment.

Another question: How do I turn OFF the pijuice so the battery will not get drained?
Thanks

tvoverbeek, 2021-08-03:
There is no direct OFF switch, but you can disable the GPIO input (if you are  powering via the RPi)
On the General tab in pijuice_cli disable 'GPIO input enable'.
Without external power connected to the PiJuice micro-USB it will as 'OFF' as you can get.

Regarding 'COMMUNICATIOn_ERROR'. Obviously the RPi does not succeed in talking to the PiJuice over the I2C bus.
Does `i2cdetect -y 1` produce the correct output? (recognized addresses 0x14 and 0x68)

GerBreOwn, 2021-08-04:
Another problem I am having is: With the RPi Zero with a PiJuice hat attached I can NOT connect to the BME280. On another RPi Zero with NO PiJuice attached I have no problems reading the BME280 sensor.

tvoverbeek, 2021-08-04:
@GerBreOwn  Looks like i2c is not enabled (BME280 also uses i2c, but depending on the breakout could also use SPI)
What is the output of `ls -l /dev/i2c*`?
If there is none, then i2c is definitely not enabled.
Use raspi-config ( `sudo raspi-config`) to enable it.

tvoverbeek, 2021-08-04:
@FaychuDon
 **Do not run pijuice_cli with sudo**
I get the same error message when running pijuice_cli with sudo.

GerBreOwn, 2021-08-04:
The output from the ls command is: crw-rw-rw- 1 root i2c 89, 1 Aug  4 13:04 /dev/i2c-1
I have configured it in raspi-config.

GerBreOwn, 2021-08-04:
I am now getting a few more options to work but am still getting errors on a lot of them. Wake-up alarm gives the error "unable to connect to device. COMMUNICATION ERROR". LEDS and BATTERY CONFIGURATION  both give a page full of errors ending with "AttributeError:  'BatteryProfileTab' object has no attribute 'profile_data'"

FaychuDon, 2021-08-04:
> @FaychuDon
> **Do not run pijuice_cli with sudo**
> I get the same error message when running pijuice_cli with sudo.

[image]

resolved thank you!

tvoverbeek, 2021-08-05:
@GerBreOwn So it seems i2c is enabled since /dev/i2c-1 exists.
But `i2cdetect -y 1` does not produce any recognized i2c devices.
Are any of your loose wires connected to pins 3 and 5 (the I2C pins)?
Anyway, first get i2cdetect working.

FaychuDon, 2021-08-05:
@GerBreOwn try the following which worked for me:

- raspi-config: enable I2C under interface options
- sudo usermod -a -G i2c <your username> - I do not have to run as pi, works for me under my newly created user account
- ensure the HAT is on the right way around (sounds stupid but i've seen someone mount it upside down)
- add the spacers on each corner when mounting HAT to pi board, at first I had them loose for testing and had similar issues to you. after mounting the HAT correctly and tightly, I was able to run pijuice (non sudo) without issue.

GerBreOwn, 2021-08-08:
GOOD NEWS. I reinstalled the PiJuice on another RPi Zero computer and now the PiJuice_cli program works as advertised.
NOW. Another question. I have seen somewhere there is software that will monitor the battery and shut down the RPi when the battery gets low. Do you know where I can get this software?
Thanks.

tvoverbeek, 2021-08-08:
@GerBreOwn Read the README on the github software page. It is all there.

triple-it, 2022-04-12:
I have the same issue...

```
pi@raspberrypi:~ $ ls /dev/i2c-1
ls: cannot access '/dev/i2c-1': No such file or directory
pi@raspberrypi:~ $ ls /dev/i2c-11
/dev/i2c-11
pi@raspberrypi:~ $ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~ $ 
```

(Note: I have also a HyperPixel2r [round lcd] connected)

tvoverbeek, 2022-04-13:
@triple-it The Hyperpixel uses almost all GPIOs including the pins for i2c-1. So you cannot put the PiJuice on the 40-pin header.
I hope you read the section "Using the alternate I2C interface on HyperPixel 4.0 (for advanced users)" in https://learn.pimoroni.com/article/getting-started-with-hyperpixel-4
Since nowadays the software i2c uses bus 11 you will have to use `sudo ln -s /dev/i2c-11 /dev/i2c-1` and make the right connections between the PiJuice and the 5-pin connector on the back of the Hyperpixel.
Also it seems you do not have the smbus module installed. install the python3-smbus module.
The PiJuice software is python3 only.

triple-it, 2022-04-13:
Hi! Thnx for the pointers @tvoverbeek !

So I would think if we:
- Dismantle the HyperPixel2r
- Run: sudo ln -s /dev/i2c-11 /dev/i2c-1
- Run: sudo systemctl start pijuice
- Delete from /boot/config.txt HyperPixel2r config
```
~~
#dtoverlay=hyperpixel2r
#enable_dpi_lcd=1
#dpi_group=2
#dpi_mode=87
#dpi_output_format=0x7f216
#dpi_timings=480 0 10 16 55 480 0 15 60 15 0 0 0 60 0 19200000 6
~~

```
It would do a bit more?

yes indeed!
Status, User_leds, etc now all seems to be working...

What is still a bit unclear now for me, is if we could get the HyperPixel2r working together with the Pijuice zero at all.
Just like the advanced descriptions on the HyperPixel4....
(Hyperpixel2r does have an extra GND, INT, SCL, SDA and 3v3 pin at the back)

Maybe someone from pijuice would know if HyperPixel2R with Pijuice Zero is even possible, and worth trying?
Then we could write the Wiki Guide for it...