PiSupply/PiJuice #768: information request: type of battery connector on the PiJuice Zero

jszigetvari, 2021-08-08:
Hello,

My only question would be what is the exact type of the battery connector on the PiJuice Zero.
I found the information that it is a JST connector, but that information is far too vague.
There are several types of JST connectors, all of which are (slightly) different.
What I know for sure is that it isn't JST-XH. It may be JST-ZH, but I couldn't yet test it, and so I am unsure.
Is it maybe something else?

Thanks for your help!

Best Regards,
János Szigetvári

tvoverbeek, 2021-08-08:
See https://github.com/PiSupply/PiJuice/tree/master/Hardware#top-of-the-board "J2 header on the PiJuice Zero"

jszigetvari, 2021-08-08:
That information is unfortunately insufficient for me.
I got a connector and wires along the PiJuice Zero board, and I would like to connect a Samsung ICR18650 rechargeable Li-ion battery, that has only two terminals.
I don't want to connect the factory connector's wires with the battery's wires in the "middle", because that's not really elegant.
I would like to buy the correct terminal type that can be put into the factory connector, and crimp two of those terminals to the ends of the battery's wires, and install them to the female connector, and then be able to easily connect my battery to the board.
That's why I need to know whether I've ordered the correct connector and crimping tool.

Thanks!

tvoverbeek, 2021-08-08:
Sorry, I cannot help you any further. Check with e.g. Digi-Key for appropriate connectors.

jszigetvari, 2021-08-08:
No problem. Anyway, thank you so much Ton for trying to help me out.

mmilann, 2021-08-09:
This is on board part https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/292253-3/1855245
And mating cable part for crimping: https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/2-179694-3/2259036

closed 2021-08-09 by mmilann

jszigetvari, 2021-08-09:
@mmilann Thank you so much for looking this up for me!

tvoverbeek, 2021-08-09:
@mmilann @jszigetvari his is the same conector info as I already gave above from the github Hardware README:
https://github.com/PiSupply/PiJuice/issues/768#issuecomment-894847441