PiSupply/PiJuice #769: PiJuice Zero + Pi4

track02, 2021-08-09:
Hello,

We're currently using regular PiJuices with Pi4's without issue however we're having difficulty getting hold of any additional PiJuice units as they seem completely sold out!

Is the PiJuice Zero compatible with the Pi4?  

We don't require long battery uptime, just enough to perform a safe shutdown when mains power is disconnected or lost

tvoverbeek, 2021-08-09:
Yes, you can use a PiJuice Zero with a Pi4.
However you cannot use the spacers for obvious reasons, so the only mechanical connection is the 40 pin GPIO connector.
ALso the battery will have to be external and connected to the PiJuice Zero battery connector.
Firmware is identical to the HAT one. However since it is not a HAT, there is no EEPROM and hence no automatic loading
of the RTC module for the emulated DS1339.

track02, 2021-08-10:
Thanks for the quick response, will order a few in to test

If I were to configure a Pi4 with a PiJuice (system events, SW1 press) and then swap the PiJuice for the PiJuice Zero would the config carry over? Assuming it'll just ignore anything not relevant (SW2, SW3)?

tvoverbeek, 2021-08-10:
Only the settings which are stored in the JSON file. Remaining settings are in the non-volatile memory of the PiJuice HAT/Zero.
See https://github.com/PiSupply/PiJuice/tree/master/Software/Source/Utilities which can be used to transfer settings.

track02, 2021-08-17:
Thank you - been testing this today and looks like its almost a drop in replacement

The only thing I think I had to change in the config were the actions for SW1 press/hold are the button bindings not saved in the json?

tvoverbeek, 2021-08-17:
Correct, they are saved in the PiJuice MCU non volatile memory.

track02, 2021-08-17:
Ah okay, we tend to work with a single master image that gets copied to all the RPi's
 I guess we'll need to do a little bit of manual config to setup the buttons for the time being

Looking at the util tool though don't think it'd be much work to add a startup script to the master image that could query the PiJuice Zero and setup the buttons automatically

tvoverbeek, 2021-08-17:
Yes, that is what I would do.

closed 2021-08-17 by track02