PiSupply/PiJuice #770: PiJuice reliable script for wake up

Jan36Nikon, 2021-08-10:
Good morning,

I want to make a device which will be power efficient and turn on only few times per day for 3 minutes do its work and shut down.

Problem which I have is that I am still not 100% sure that PiJuice will start on specified time.
I have spent last 5 days trying to get the information from Githib Issues. But there is no straightforward guide.

-----------------------
My setup is:
Raspberry Pi Zero with latest Raspbian
PiJuice Zero with latest firmware
Power connected to PiJuice Zero
No battery, will be in future...

Device will have internet connection, but I want from it to work also without the Internet connection

Address for HAT 50
in config.txt I have dtoverlay=i2c-rtc,ds1339=1
In i2cdetect -y1 I am getting UU in position 68
-----------------------

I used this post #91 which helped me a lot.
https://github.com/PiSupply/PiJuice/issues/91#issuecomment-381324352

This current script worked for me with no problem for 24 hours.

But when I update few parameters: DELTA_MIN=15 and lenght of fake job for 180 sec

Result was that after few hours device was stucked in power off and no wakeup occured, more info on this log file:
My Timezone is UTC + 2
Boot of Raspberry takes aproximately 1 minute because of USB mounted drive.

```
2021-08-09 19:31:24 -- Started
2021-08-09 19:50:04 -- Started
2021-08-09 20:09:04 -- Started
2021-08-09 20:28:05 -- Started
2021-08-09 20:47:04 -- Started
2021-08-09 21:06:03 -- Started
2021-08-09 21:25:05 -- Started
2021-08-09 21:44:04 -- Started
2021-08-09 22:03:03 -- Started
2021-08-09 22:22:03 -- Started
2021-08-09 22:41:05 -- Started
2021-08-09 23:00:04 -- Started
2021-08-09 23:19:05 -- Started
2021-08-09 23:38:04 -- Started
2021-08-09 23:57:04 -- Started
2021-08-10 00:16:01 -- Started
**2000-01-01** 02:32:52 -- Started
```

Could you please help me to make it reliable?
Thank you very much,
Jan

My current code:

```
from __future__ import print_function

import time
import pijuice
import subprocess
import datetime
import os
import sys 

DELTA_MIN=15

# Rely on RTC to keep the time
subprocess.call(["sudo", "hwclock", "--hctosys"])

# Record start time
txt = datetime.datetime.now().strftime('%Y-%m-%d %H:%M:%S') + ' -- Started\n'
with open('/home/pi/test.log','a') as f:
    f.write(txt)

# This script is started at reboot by cron.
# Since the start is very early in the boot sequence we wait for the i2c-1 device
while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

try:
    pj = pijuice.PiJuice(1, 0x14)
except:
    print("Cannot create pijuice object")
    sys.exit()

# Do the work
for i in range(180):
   print('*', end='')
   sys.stdout.flush()
   time.sleep(1)
print()

# Set RTC alarm 5 minutes from now
# RTC is kept in UTC
a={}
a['year'] = 'EVERY_YEAR'
a['month'] = 'EVERY_MONTH'
a['day'] = 'EVERY_DAY'
a['hour'] = 'EVERY_HOUR'
t = datetime.datetime.utcnow()
a['minute'] = (t.minute + DELTA_MIN) % 60
a['second'] = 0
status = pj.rtcAlarm.SetAlarm(a)
if status['error'] != 'NO_ERROR':
    print('Cannot set alarm\n')
    sys.exit()
else:
    print('Alarm set for ' + str(pj.rtcAlarm.GetAlarm()))

# Enable wakeup, otherwise power to the RPi will not be
# applied when the RTC alarm goes off
pj.rtcAlarm.SetWakeupEnabled(True)
time.sleep(0.4)

# PiJuice shuts down power to Rpi after 20 sec from now
# This leaves sufficient time to execute the shutdown sequence
pj.power.SetPowerOff(20)
subprocess.call(["sudo", "poweroff"])
```

tvoverbeek, 2021-08-10:
A few things.
Do no use `dtoverlay=i2c-rtc,ds1339=1` but `dtoverlay=i2c-rtc,ds1307=1`
With ds1339 the OS can clear the alarm if the time gets synchronised when connected to the internet.
The RTC also needs I2C, so before you copy the RTC time to the system the i2c bus has to be initialized.
The pijuice service startup does a check and reloads the rtc module if needed.
See lines 357-391 in pijuice_sys.py (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py).
Use the startup event hook (which gets called by the service after the RTC is available) to do the subprocess.call(["sudo", "hwclock", "--hctosys"]).
Your @reboot job could start too early (before RTC is initialized). Put in a delay at the start or do some form of synchronisation with the service startup event function.

Jan36Nikon, 2021-08-11:
Thank you for support,
I just changed dtoverlay=i2c-rtc,ds1339=1 to dtoverlay=i2c-rtc,ds1307=1
and also adjust the timing of crontab to start 60 secconds after reboot.
I am not really sure how to use service startup event. But basically this setup should work also.

`@reboot sleep 60 && /usr/bin/python3 /mnt/OGC/001_CurrentScript/Script.py`

Can I also ask if there is good aproach to use the ClearAlarmFlag on the end of the script in front of shutdown command?
So the end will look like this?

```
.........
# Enable wakeup, otherwise power to the RPi will not be
# applied when the RTC alarm goes off
pj.rtcAlarm.SetWakeupEnabled(True)
time.sleep(0.4)

# PiJuice shuts down power to Rpi after 20 sec from now
# This leaves sufficient time to execute the shutdown sequence
pj.rtcAlarm.ClearAlarmFlag()
pj.power.SetPowerOff(20)
subprocess.call(["sudo", "poweroff"])
```

I will start test for 24 hours now and let you know if it will work.
Thank you for your time,
Jan

Jan36Nikon, 2021-08-12:
It looks like the wakeup enabled was disabled by the time sync.
So according to https://github.com/PiSupply/PiJuice/issues/362
I changed address of HAT to 52 and hope it will work OK now.

I will let you knou about result.

tvoverbeek, 2021-08-12:
Above you wrote you are using a PiJuice Zero. In that case changing the EEPROM address to 0x52 has no effect
(The Pijuice Zero has no HAT EEPROM).
For the PiJuice Zero you will have to use `dtoverlay=i2c-rtc,ds1307=1)` as I already wrote above.
The DS1307 has no alarm, but the clock part is identical to the DS1339. So you get internet time synchronization
of the RTC without clearing the alarm.
Let me know how it goes.

Jan36Nikon, 2021-08-12:
Like I wrote yesterday, after setting dtoverlay=i2c-rtc,ds1307=1 I still find raspberry in off state with wakeup enable button disabled. I had script to wake up every 19 minutes. Could short time power loss be the issue in that case? 

After that I changed the script to the following on end of my post, to wake up at every hour and EEPROM address from 50 to 52 (it should have no effect).
Could you please check my script if it is OK, especially the ending with 
```
pj.rtcAlarm.ClearAlarmFlag()
pj.rtcAlarm.SetWakeupEnabled(True)
```
If they are in correct order. And if I need to use ClearAlarmFlag at all?

My current setup is:
EEPROM address 0x52 (no effect, but it is set to this address)
dtoverlay=i2c-rtc,ds1307=1
No battery connected - only micro USB connected from power outlet to PiJuice PCB

Crontab entry:
`
@reboot sleep 120 && /usr/bin/python3 /mnt/OGC/001_CurrentScript/Script.py
`

Script.py
```
from __future__ import print_function

import time
import pijuice
import subprocess
import datetime
import os
import sys 

# Rely on RTC to keep the time
subprocess.call(["sudo", "hwclock", "--hctosys"])

# Record start time
txt = datetime.datetime.now().strftime('%Y-%m-%d %H:%M:%S') + ' -- Started\n'
with open('/home/pi/test.log','a') as f:
    f.write(txt)

# This script is started at reboot by cron.
# Since the start is very early in the boot sequence we wait for the i2c-1 device
while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

try:
    pj = pijuice.PiJuice(1, 0x14)
except:
    print("Cannot create pijuice object")
    sys.exit()

# Do the work
for i in range(180):
   print('*', end='')
   sys.stdout.flush()
   time.sleep(1)
print()

# Set RTC alarm 5 minutes from now
# RTC is kept in UTC
a={}
a['year'] = 'EVERY_YEAR'
a['month'] = 'EVERY_MONTH'
a['day'] = 'EVERY_DAY'
a['hour'] = 'EVERY_HOUR'
a['minute'] = 0
a['second'] = 0
status = pj.rtcAlarm.SetAlarm(a)
if status['error'] != 'NO_ERROR':
    print('Cannot set alarm\n')
    sys.exit()
else:
    print('Alarm set for ' + str(pj.rtcAlarm.GetAlarm()))

# Enable wakeup, otherwise power to the RPi will not be
# applied when the RTC alarm goes off
pj.rtcAlarm.ClearAlarmFlag()
pj.rtcAlarm.SetWakeupEnabled(True)
time.sleep(0.4)

# PiJuice shuts down power to Rpi after 20 sec from now
# This leaves sufficient time to execute the shutdown sequence
pj.power.SetPowerOff(20)
subprocess.call(["sudo", "poweroff"])
```
Current state of log
Right now it is working for 24 hours but from last error in wakeup I changed the script only from 19 minutes interval to 1 hour interval and EEPROM address prom 50 to 52 (which should have no effect), so I am not perfectly sure that the error was solved...

Thank you for your comment,
Jan

```
2021-08-11 16:28:17 -- Started
2021-08-11 17:02:48 -- Started
2021-08-11 18:02:49 -- Started
2021-08-11 19:02:47 -- Started
2021-08-11 20:02:48 -- Started
2021-08-11 20:44:48 -- Started
2021-08-11 21:02:49 -- Started
2021-08-11 22:02:49 -- Started
2021-08-11 23:02:50 -- Started
2021-08-12 00:02:49 -- Started
2021-08-12 01:02:49 -- Started
2021-08-12 02:02:53 -- Started
2021-08-12 03:02:46 -- Started
2021-08-12 04:02:48 -- Started
2021-08-12 05:02:50 -- Started
2021-08-12 06:02:49 -- Started
2021-08-12 07:02:48 -- Started
2021-08-12 08:02:50 -- Started
2021-08-12 09:02:50 -- Started
2021-08-12 10:02:46 -- Started
2021-08-12 11:02:50 -- Started
2021-08-12 12:02:50 -- Started
2021-08-12 13:02:47 -- Started
2021-08-12 14:02:47 -- Started
2021-08-12 15:02:52 -- Started
2021-08-12 16:02:47 -- Started
```

tvoverbeek, 2021-08-12:
A few things about your script.
- As it is executed by python3 you do not need the `from __future__ import ...` line
- Move the RTC to system clock (`subprocess.call(["sudo", "hwclock", "--hctosys"])`) after the check for `/dev/i2c-1`
  since the hwclock (= the DS1307 in the PiJuice) needs i2c working.
- In Do the work you can change the print statement to `print('*', end='', flush=True)` and delete the `sys.stdout.flush()` line.

The script was probably written for python2.
The remainder of the  logic looks fine.

Jan36Nikon, 2021-08-13:
Thank you for your update,
I just check the process and it looks like it again stuck in power off mode. 

See the log:

```
2021-08-11 16:28:17 -- Started
2021-08-11 17:02:48 -- Started
2021-08-11 18:02:49 -- Started
2021-08-11 19:02:47 -- Started
2021-08-11 20:02:48 -- Started
2021-08-11 20:44:48 -- Started
2021-08-11 21:02:49 -- Started
2021-08-11 22:02:49 -- Started
2021-08-11 23:02:50 -- Started
2021-08-12 00:02:49 -- Started
2021-08-12 01:02:49 -- Started
2021-08-12 02:02:53 -- Started
2021-08-12 03:02:46 -- Started
2021-08-12 04:02:48 -- Started
2021-08-12 05:02:50 -- Started
2021-08-12 06:02:49 -- Started
2021-08-12 07:02:48 -- Started
2021-08-12 08:02:50 -- Started
2021-08-12 09:02:50 -- Started
2021-08-12 10:02:46 -- Started
2021-08-12 11:02:50 -- Started
2021-08-12 12:02:50 -- Started
2021-08-12 13:02:47 -- Started
2021-08-12 14:02:47 -- Started
2021-08-12 15:02:52 -- Started
2021-08-12 16:02:47 -- Started
2021-08-12 17:02:49 -- Started
2021-08-12 18:02:50 -- Started
2021-08-12 19:02:49 -- Started
2021-08-12 20:02:52 -- Started
2021-08-12 21:02:50 -- Started
2021-08-12 22:02:46 -- Started
2021-08-12 23:02:47 -- Started
2021-08-13 00:02:46 -- Started
2021-08-13 01:02:46 -- Started
```
I found it again after several hours in OFF state.

When I displayed PiJuice Zero GUI after manual boot I found this: (disabled "WakeUp enabled")

[image]

I will try to put your modification in the script so it will look like this and I will again start the test:

Is it necessary to have 
`time.sleep(0.4) `
after 
`pj.rtcAlarm.SetWakeupEnabled(True)`

Thank you for support.
Jan

```
import time
import pijuice
import subprocess
import datetime
import os
import sys 

# Record start time
txt = datetime.datetime.now().strftime('%Y-%m-%d %H:%M:%S') + ' -- Started\n'
with open('/home/pi/test.log','a') as f:
    f.write(txt)

# This script is started at reboot by cron.
# Since the start is very early in the boot sequence we wait for the i2c-1 device
while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

try:
    pj = pijuice.PiJuice(1, 0x14)
    # Rely on RTC to keep the time
    subprocess.call(["sudo", "hwclock", "--hctosys"])
except:
    print("Cannot create pijuice object")
    sys.exit()

# Do the work
for i in range(180):
    print('*', end='', flush=True)
    time.sleep(1)
print()

# Set RTC alarm 5 minutes from now
# RTC is kept in UTC
a={}
a['year'] = 'EVERY_YEAR'
a['month'] = 'EVERY_MONTH'
a['day'] = 'EVERY_DAY'
a['hour'] = 'EVERY_HOUR'
a['minute'] = 0
a['second'] = 0
status = pj.rtcAlarm.SetAlarm(a)
if status['error'] != 'NO_ERROR':
    print('Cannot set alarm\n')
    sys.exit()
else:
    print('Alarm set for ' + str(pj.rtcAlarm.GetAlarm()))

# Enable wakeup, otherwise power to the RPi will not be
# applied when the RTC alarm goes off
pj.rtcAlarm.ClearAlarmFlag()
pj.rtcAlarm.SetWakeupEnabled(True)
time.sleep(0.4)

# PiJuice shuts down power to Rpi after 20 sec from now
# This leaves sufficient time to execute the shutdown sequence
pj.power.SetPowerOff(20)
subprocess.call(["sudo", "poweroff"])
```

tvoverbeek, 2021-08-13:
It smells like the wakeup at 02:00 actually happened, but due to insufficient power the Pi startup did not succeed and hence the wakeup was left disabled.
Do you have an external disk connected since your script is running from /mnt/OGC?
To get a clue can you read and log the charge status (pj.status.GetChargeLevel()) and battery voltage (pj.status.GetBatteryVoltage()) in your Do the work?

Jan36Nikon, 2021-08-13:
Yes, I have USB flash drive 128 Gb mounted, but it is not any HDD, only flashdrive.
I will try to add those commands to code and we will see.

Thank you for your support,
Jan

Jan36Nikon, 2021-08-13:
Thank you for your support, important point is that I donť have any battery connected...

Yes, I have USB flash drive 128 Gb mounted, but it is not any HDD, only flashdrive.
I will try to add those commands to the code and we will see.

Thank you for your support,
Jan

tvoverbeek, 2021-08-13:
On the HAT configuration tab make sure you have 'no battery turn on' enabled when you are running without battery.
Also on your screenshot above you have Wakeup enabled: Last 08:03:16
This means the alarm flag was still set when you started the GUI around 08:03. So the alarm went off but the flag was not cleared.
USB flash drive consumption can be an extra 200mA, but depends on the type of flash drive.

Jan36Nikon, 2021-08-13:
In configuration of HAT I enable 'no battery turn on' previously it was disabled.

How can I be sure that alarm flag will be cleared?
I also changed the 230V power adapter, and we will see...

 'no battery turn on'  button has big affect on system function?

Thank you,
Jan

tvoverbeek, 2021-08-13:
From the software README:
No battery turn on: If enabled pijuice will automatically power on 5V rail and trigger wake-up as soon as power appears at USB Micro Input and there is no battery. Disabled by default.

Jan36Nikon, 2021-08-14:
Thank you I will let you know how it works...

Jan36Nikon, 2021-08-15:
It looks like your last tip according to 'No battery turn on' solved the problem, right now it is working for 2 days in row.

- Yes that radiobutton solved the problem, but could you please explain me wat was the problem?
- In log you can see that sometimes device boot also in time when it should sleep:
(7:54:09, 17:46:21) do you have any explanation for it (Isn't it the time when previously device was left in power off state?)

```
20:02:50
21:02:47
22:02:49
23:02:47
0:02:46
1:02:46
2:02:48
3:02:49
4:02:47
5:02:49
6:02:47
7:02:47
7:54:09
8:02:46
9:02:46
10:02:48
11:02:49
12:02:50
13:02:50
14:02:49
15:02:45
16:02:50
17:02:49
17:46:21
18:02:47
19:02:50
20:02:46
21:02:47
22:02:52
23:02:49
0:02:45
1:02:50
2:02:50
3:02:50
4:02:47
5:02:47
6:02:47
7:02:49
8:02:50
9:02:49
10:02:46
11:02:47
12:02:46
```

Thank you for support,
Jan

Jan36Nikon, 2021-08-15:
Interesting is that hour when device turn on twice is the same hour when I connected to it through VNC (at the first 2 minutes of that hour)

Jan36Nikon, 2021-08-16:
Another update from my new test...

Right now device is not failing to start...

But in log you can see that it is also waking up in different times than every hour...
Do you know kow to solve it?

Thanks,
Jan

```
2021-08-15 16:55:11 -- Started
2021-08-15 17:02:50 -- Started
2021-08-15 18:02:47 -- Started
2021-08-15 19:02:49 -- Started
2021-08-15 20:02:48 -- Started
2021-08-15 21:02:49 -- Started
2021-08-15 21:18:16 -- Started
2021-08-15 22:02:46 -- Started
2021-08-15 23:02:47 -- Started
2021-08-15 23:29:37 -- Started
2021-08-16 00:02:49 -- Started
2021-08-16 00:08:33 -- Started
2021-08-16 02:02:46 -- Started
2021-08-16 03:02:48 -- Started
2021-08-16 03:52:15 -- Started
2021-08-16 04:02:48 -- Started
2021-08-16 05:02:50 -- Started
2021-08-16 06:02:47 -- Started
2021-08-16 07:02:47 -- Started
2021-08-16 08:02:48 -- Started
2021-08-16 09:02:45 -- Started
2021-08-16 10:02:45 -- Started
2021-08-16 11:02:50 -- Started
2021-08-16 12:02:46 -- Started
2021-08-16 13:02:46 -- Started
2021-08-16 14:02:49 -- Started
2021-08-16 14:13:01 -- Started
2021-08-16 15:02:49 -- Started
2021-08-16 16:02:47 -- Started
2021-08-16 16:44:31 -- Started
2021-08-16 17:02:48 -- Started
2021-08-16 18:02:46 -- Started
```

tvoverbeek, 2021-08-17:
Since you have 'no battery turn on' enabled, also any power interruption on the PiJuice micro-USB input will also wake-up the pi. Could that have happened? Power brown-outs due to bad weather?

Jan36Nikon, 2021-08-17:
Thank you for tips I will do another tests.

Jan36Nikon, 2021-09-12:
Good afternoon,

I did a test and everything was working OK without battery.

- Now I purchased also the Battery 5000 mAh original PiJuice and changed the settings of the HAT to know that battery is present.

Currently I am getting same problem as in issue #771 (run without power source only on battery)
```
hwclock: Cannot access the Hardware Clock via any known method.
hwclock: Use the --verbose option to see the details of our search for an access method.
```
Obviously it looks like problem is in this row:
`subprocess.call(["sudo", "hwclock", "--hctosys"])`

**Is it neccessar only when I am running without Internet connection?**

According to LOG, Image was taken and also next alarm was set successfully.
I am trying to run next test with force_turbo=1 in config.txt

It could be caused by USB flash drive which I am using to store my captured images, like you told in Issue #771 
(I am trying not to store images on SD card, is there any other possibility?)

```
2021-09-10 21:00:18 | Image OK | Alarm OK | Battery level: 93%
2021-09-10 22:00:23 | Image OK | Alarm OK | Battery level: 93%
2021-09-10 23:00:19 | Image OK | Alarm OK | Battery level: 92%
2021-09-11 00:00:22 | Image OK | Alarm OK | Battery level: 92%
hwclock: Cannot access the Hardware Clock via any known method.
hwclock: Use the --verbose option to see the details of our search for an access method.
2021-09-11 01:00:20 | Image OK | Alarm OK | Battery level: 91%
2021-09-11 02:00:21 | Image OK | Alarm OK | Battery level: 90%
2021-09-11 03:00:23 | Image OK | Alarm OK | Battery level: 90%
```

Current Script:
```
import pijuice, time, os
import subprocess
import sys
from time import sleep
from datetime import datetime
from picamera import PiCamera

def CheckTime():
    while not os.path.exists('/dev/i2c-1'):
        time.sleep(0.1)

    pj = pijuice.PiJuice(1, 0x14)
    subprocess.call(["sudo", "hwclock", "--hctosys"])

    return pj

def ImageCapture():
    try:
        ImageCaptureState = 'Image NoINFO'
        camera = PiCamera()
        camera.resolution = (4056, 3040)
        #FullResolution = 4056 x 3040
        camera.framerate = 30
        camera.shutter_speed = 33333
        camera.iso = 100
        camera.start_preview(resolution=(400, 300))
        time.sleep(30)
        camera.exposure_mode = 'off'
        outfile = "/mnt/OGC/RAW/%s.jpg" % (datetime.now().strftime("%Y%m%d %H%M%S"))
        camera.capture(outfile)
        camera.close()
        ImageCaptureState = 'Image OK'
        return ImageCaptureState
    except Exception as err:
        ImageCaptureState = 'Image ERROR'
        return ImageCaptureState
  
def setAutomaticWakeUP(pj, ImageCaptureState):
    a={}
    a['year'] = 'EVERY_YEAR'
    a['month'] = 'EVERY_MONTH'
    a['day'] = 'EVERY_DAY'
    a['hour'] = 'EVERY_HOUR'
    a['minute'] = 57
    a['second'] = 00
    status = pj.rtcAlarm.SetAlarm(a)
    battery = pj.status.GetChargeLevel()
    if status['error'] != 'NO_ERROR':
        print(datetime.now().strftime("%Y-%m-%d %H:%M:%S") + " | " + ImageCaptureState + " | Alarm ERROR | Battery level: " + str(battery['data']) + "%")
        sys.exit()
    else:
        print(datetime.now().strftime("%Y-%m-%d %H:%M:%S") + " | " + ImageCaptureState + " | Alarm OK | Battery level: " + str(battery['data']) + "%")
       
def TurnOff(pj):
    pj.rtcAlarm.ClearAlarmFlag()
    pj.rtcAlarm.SetWakeupEnabled(True)
    time.sleep(0.4)
    pj.power.SetPowerOff(20)
    subprocess.call(["sudo", "poweroff"])

pj = CheckTime()
sleep(1)
ImageCaptureState = ImageCapture()
sleep(1)
setAutomaticWakeUP(pj, ImageCaptureState)
sleep(1)
TurnOff(pj)

exit()
```

tvoverbeek, 2021-09-13:
If you have an internet connection you do not need the line: `subprocess.call(["sudo", "hwclock", "--hctosys"])`.
The timesyncd will sync the system time from the internet.
Also with an internet connection you could store your picture in a tmpfs location, e.g. /run (=ram storage)
and copy/store it on another computer using e.g. scp or samba (smb mounts) and delete it after successful copy.

Jan36Nikon, 2021-09-14:
Thank you for your quick answer.
Point is that I don't have access to internet all the time.

Right now I removed this row from my script.
`subprocess.call(["sudo", "hwclock", "--hctosys"]).`

And it is working OK without the internet on the battery.

I think that because I am starting Crontab 2 minutes after boot, system have plenty of time to synchronize time from RTC to system clock.