PiSupply/PiJuice #771: PiJuice Zero - hwclock Remote I/O error / can not access the Hardware Clock

ingelumer, 2021-08-11:
Hello,

I guess my issue is not totally new, but I haven't found a working solution going through the existing Posts.

The issue is repeatable, I have just set up a new PiZero Installation, which started good, but then ran into the same issues.
My Pi Zero is connect to the Pijuice Zero board and running on battery power.

After some reboots I noticed following error message when running hwclock -r :
`hwclock: ioctl(4, RTC_UIE_ON, 0) to /dev/rtc0 failed: Remote I/O error`

I rebooted again and started to receive constantly below error:
hwclock: Cannot access the Hardware Clock via any known method.

```
sudo hwclock -v
hwclock from util-linux 2.33.1
System Time: 1628677018.998845
Trying to open: /dev/rtc0
Trying to open: /dev/rtc
Trying to open: /dev/misc/rtc
No usable clock interface found.
hwclock: Cannot access the Hardware Clock via any known method.
```

Also I am getting Communication Errors when instantiating PiJuice, usually it takes something like 4 tries until success.
This worked fine just before receiving the hwclock error.

Output of i2cdetect -y 1 : 
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

pijuice_cli is running fine with EEPROM 52 / I2C address 14 / I2C RTC 68
Firmware is uptodate, v1.5

/boot/config.txt:
`dtoverlay=i2c-rtc,ds1339=1`

Any help is appreciated.

tvoverbeek, 2021-08-11:
RTC is only available when the i2c bus is initialized.
If you try to access it too early (i.e. from a cron @reboor job or from /etc/rc.local) you get this type of errors.
See also #770 for some more info.
Getting communication error also indicated problems on the i2c bus.
Are you running jobs with varying cpu load? Then the pi may change its clock frequency depending on the load.
This also affects the i2c clock. A varying i2c clock during an i2c transaction also gives this type of errors.
You can try to add `force_turbo=1` to config.txt which causes the clock to stay constant.

ingelumer, 2021-08-11:
Thanks for your quick feedback.
Indeed I am using crontab to run a job that uses pijuice to schedule the next alarm.
I disabled the cron entry for my new test runs, but the error still appears.
My job not should not cause heavy cpu load. Adding the turbo entry does not make a difference for me.
Also changing the other overlay to i2c-rtc,ds1307=1 still shows the same error for hwclock.

You mentioned the startup event hook method, how can I apply it to my system?

At last I have detected an interesting behavior.
The issue with hwclock is only happening when booting on Battery Power.
With Power through PiJuice USB, hwclock works just right.
Does this help?

ingelumer, 2021-08-13:
I performed some more test cycles, started the system with USB and/or Battery only.
And clearly the issue of the hwclock error only happening when connected solely to battery power.
There the hwclock command and PiJuice initialization give random errors.
Error Message: hwclock: ioctl(4, RTC_UIE_ON, 0) to /dev/rtc0 failed: Remote I/O error

As soon as USB cable has been plugged / or rebooted with, there is no error at all.

Very seldom I saw the error: hwclock: Cannot access the Hardware Clock via any known method.

The batteries/akkus I tested were both from PiJuice, 5000 and 10000 mAH.

Could the hwclock issue when running on battery be related to the PiJuice Zero hardware?

tvoverbeek, 2021-08-13:
What else is connected to your Pi Zero? It smells like an unstable 5v power problem.
For instance if you have an external usb disk connected, this will draw excessive power when starting.

ingelumer, 2021-08-13:
There is no other device attached, just PiZero, PiJuice Zero, PiJuice Battery

tvoverbeek, 2021-08-13:
The startup event shows up on the system events tab on the GUI/CLI.
You can attach a user function there. Note this function is executed as part of the pijuice service startup, so it should be relatively short (no endless loops please). You need the latest software and firmware versions for using the startup event.

ingelumer, 2021-08-16:
Sorry I am not clear on your last statement, do you want me to add the hwclock --hctosys to the PiJuice Startup event, or my script that I start through crontab?
like you commented in #[770](https://github.com/PiSupply/PiJuice/issues/770#issuecomment-896300511)?

What is strange too that I had several error message in the PiJuice, mostly while saving settings.
eg Can not connect to device, Failed to communicate with PiJuice Service...

tvoverbeek, 2021-08-17:
If you get regular error messages then your I2C communication is flaky for whatever reason (bad connections, bad power, ...).
The suggestion to use the statup event was to make sure the I2C is initialized before you try to set the system clock from the RTC. Using a sufficient delay in your cron job should do the same thing.