PiSupply/PiJuice #783: Not in Bullseye Repos

ph818, 2021-09-28:
Hello,

Would it be possible to put this package in the bullseye repos?

ph818, 2021-09-28:
When I install it manually, and try to run pijuice_cli, it does this

```
$ pijuice_cli
/usr/bin/pijuice_cli.py:116: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if default is not None and default is not "":
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2005, in <module>
    current_fw_version = get_current_fw_version()
  File "/usr/bin/pijuice_cli.py", line 143, in get_current_fw_version
    status = pijuice.config.GetFirmwareVersion()
  File "/usr/lib/python3/dist-packages/pijuice.py", line 1535, in GetFirmwareVersion
    ret = self.interface.ReadData(self.FIRMWARE_VERSION_CMD, 2)
  File "/usr/lib/python3/dist-packages/pijuice.py", line 92, in ReadData
    if not self._DoTransfer(self._Read):
  File "/usr/lib/python3/dist-packages/pijuice.py", line 79, in _DoTransfer
    while self.t.isAlive() and n < 2:
AttributeError: 'Thread' object has no attribute 'isAlive'
```

Jogib, 2021-11-09:
When can we expect to see the pijuice package in the bullseye repo?
Can't even download and install the DEB file manually, since it's not recognized as a debian package.

tvoverbeek, 2021-11-10:
Cannot give you a time table. First the dust has to settle (i.e. problems with recognizing HDMI displays, VNC does not work yet, bugs in screen configuration utility)
Then there is the change to a later python version, and gtk2 to gtk3, change of window manager which affects pijuice-gui.
It is on my todo list, but I am doing this as a volunteer and hence I cannot give you a specific date.

corwinou, 2021-11-11:
same issue for me !!! :o(

tvoverbeek, 2021-11-11:
LLook at what you have to change to get it running under Bullseye in issue #784 at least forthe command line part.

Jogib, 2021-11-13:
Got it to work in Bullseye.
Turned out, just doing a wget to download the 1.6 package made the DEB file corrupt. Dunno why.

This is what i did:
Installed git, and cloned the whole repo.
Ran the 1.6 deb package from pijuice/software/install
Install failed, because of missing dependencies. Installed those and ran the 1.6 deb package again.

Altered the code in the 2 files from issue #784 
Could now run "pijuice_cli". Though some data was all wrong.
 
Ran the 1.7 installer, and now everything seems to be in order.

ph818, 2023-02-04:
Recently installed pijuice-base on Raspios64, so I am closing this issue.

closed 2023-02-04 by ph818