PiSupply/PiJuice #784: Troubleshooting the hardware

ph818, 2021-09-28:
Hello,

I got a PiJuice "used" from a large online retailer to use as a UPS; I am having some problems and wondering if there is a way to troubleshoot what "should" be happening.

If I plug microUSB into the PiJuice, I get the blue "charging" LED and the red "battery low/absent" LED. I am expecting the Red LED to change color as it charges, but it stays red. It provides power to the Pi, but I experienced interruptions over SSH. When I unplugged from the mains, the pi immediately lost power.

If I plug the microUSB into the pi, the blue "charging" LED turns on once every ~2 seconds, and the battery LED starts red and dim, gets red and bright, and then turns off.

My PiJuice arrived with the battery brackets disassembled and [one missing](https://imgur.com/a/h8elNZh). The battery contacts are secure though, so I went ahead and installed it onto the Pi.

I was not able to install the software with apt; I am on raspbian bullseye, but looking at the debian repositories, I can't find it there either. I manually installed the 1.6 version from github, but it failed with a traceback.

```
$ pijuice_cli
/usr/bin/pijuice_cli.py:116: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if default is not None and default is not "":
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2005, in <module>
    current_fw_version = get_current_fw_version()
  File "/usr/bin/pijuice_cli.py", line 143, in get_current_fw_version
    status = pijuice.config.GetFirmwareVersion()
  File "/usr/lib/python3/dist-packages/pijuice.py", line 1535, in GetFirmwareVersion
    ret = self.interface.ReadData(self.FIRMWARE_VERSION_CMD, 2)
  File "/usr/lib/python3/dist-packages/pijuice.py", line 92, in ReadData
    if not self._DoTransfer(self._Read):
  File "/usr/lib/python3/dist-packages/pijuice.py", line 79, in _DoTransfer
    while self.t.isAlive() and n < 2:
AttributeError: 'Thread' object has no attribute 'isAlive'
```

I was able to run i2cdetect, and got these results. (I made sure to turn on I2C in raspi-config prior).

```
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Any help or advice would be appreciated. Thank you.

tvoverbeek, 2021-09-28:
In your copy of /usr/bin/pijuice_cli.py change line 116 from
`if default is not None and default is not "":`
to:
`if default is not None and default != "":`
This is because Bullseye uses a later version of python3 than Buster.
.
This is already fixed on the github version. See issue #498
With a working pijuice_cli you should be able to get more info about the status of your PiJuice and battery

ph818, 2021-09-28:
I installed version 1.6 from github, are you talking about version 1.7?

tvoverbeek, 2021-09-28:
There is no official 1.7 version yet. The github source includes fixes and new features since 1.6 which will be included when 1.7 is released.
For now just manually update your copy of /usr/bin/pijuice_cli.py.

ph818, 2021-09-28:
Thank you. I manually updated it, and I get this traceback now:

```
$ pijuice_cli
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2005, in <module>
    current_fw_version = get_current_fw_version()
  File "/usr/bin/pijuice_cli.py", line 143, in get_current_fw_version
    status = pijuice.config.GetFirmwareVersion()
  File "/usr/lib/python3/dist-packages/pijuice.py", line 1535, in GetFirmwareVersion
    ret = self.interface.ReadData(self.FIRMWARE_VERSION_CMD, 2)
  File "/usr/lib/python3/dist-packages/pijuice.py", line 92, in ReadData
    if not self._DoTransfer(self._Read):
  File "/usr/lib/python3/dist-packages/pijuice.py", line 79, in _DoTransfer
    while self.t.isAlive() and n < 2:
AttributeError: 'Thread' object has no attribute 'isAlive'
```

tvoverbeek, 2021-09-28:
OK. Next change needed due to python3.9 in Bullseye:

In /usr/lib/python3/dist-packages/pijuice.py in function _DoTransfer (lines 70-85)
change the  three occurrences of `isAlive()` to `is_alive()`

As you experienced no testing has been done on Bullseye yet.

ph818, 2021-09-28:
Thanks! It is up and running now.

In status, I see that the battery is:
* 0%
* 0.260V (fluctuates)
* 65320 deg.C
* "NOT_PRESENT"

GPIO pins show
* 5.05 V input (fluctuates)
* 3.464 A
* NOT_PRESENT

USB Micro Power Input: present
Fault: none
System Switch: off

ph818, 2021-09-30:
If I plug power into the raspberry pi directly, I get these values:
* Battery: 0%, 0.737V, 65319 degC, NOT_PRESENT
* GPIO Power Input: 5.367 V, 0.000A, NOT_PRESENT
* USB Micro Power: not present.

Are these values expected, or is there a fault in the hardware? I'm particularly interested in the high amperage read when no battery is read in the first case, and confused by the non-zero "battery voltage" when no battery is present.

The temperature reading I can live with :)

tvoverbeek, 2021-09-30:
FIrst case: GPIO current flows from the PiJuice to the RPi.
If the RPi is a Pi4 with some peripherals 3.4A sounds reasonable.
Note: GPIO current is not actually measured but estimated.
The other way around there is (almost) no current from the GPIO to the PiJuice.
Could be right, since there is no battery connected.
Probably in the settings battery temperature sense is set to NTC. SInce it is open circuit you get a high temperature.

ph818, 2021-09-30:
It is a 3B+ with nothing connected but ethernet; Would it still be
reasonable?

The questions I am trying to answer:
* What is going on with the battery?
* Is the PiJuice killing batteries or is it working as normal

LeighdePaor, 2021-12-09:
I am also using a 3B+ and have one of the older PiJuice hats, I have downloaded version 1.7 using : 
wget https://github.com/PiSupply/PiJuice/raw/master/Software/Install/pijuice-base_1.7_all.deb
I get the following error when I install run the pijuice_cli and select "general" it reports "no device detected"
When I select "status" the unit faults with a traceback:
+++++++++++++++++++++++++++++++++++++++
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2074, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 287, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 385, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 790, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 827, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 416, in <lambda>
    wrapper = lambda: self.parse_input(
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 515, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 412, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 513, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 598, in keypress
    return self.top_w.keypress(self.top_w_size(size,
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1626, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2316, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 968, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 543, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 461, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1981, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 209, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 269, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 224, in get_status
    general_info += ", %3d°C, %s" % (temp, status['battery'])
KeyError: 'battery'

+++++++++++++++++++++++++++++++++++++++

If I run pijuice_cli using sudo it faults out with a traceback on selection of "general"

+++++++++++++++++++++++++++++++++++++++
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2074, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 287, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 385, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 790, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 827, in _loop
    self._watch_files[fd]()
[... 15 lines cut ...]
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 968, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 543, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 461, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1981, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 209, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 269, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 215, in get_status
    status = pijuice.status.GetStatus().get('data', {})
AttributeError: 'NoneType' object has no attribute 'status'

+++++++++++++++++++++++++++++++++++++++
Currently running raspbian bullseye and Python 3.9.2

LeighdePaor, 2021-12-09:
> Is I2C enabled? How did you install the package?

Thanks for getting back so fast, I ended up cloning your repo and walking through the README.md files, ran the pyjuicetest.py and aw I had a VERY old firmware. Followed the instructions for pijuiceboot and applied the latest firmware and now everything is hunky-dory.

Error resolved by firmware update.

tvoverbeek, 2021-12-09:
OK. Good to know you got it working.
Closing the issue.

closed 2021-12-09 by tvoverbeek