PiSupply/PiJuice #785: User Function wont run

kymadn, 2021-10-02:
Hello word, I have been struggling to get a user function script to run from my PiJuice. 

I am testing with a file /home/pi/buttonlog.py

I have also made this file executable:
sudo chmod 777 /home/pi/buttonlog.py

[image]

My current button assignment SW2:
[image]

I have also set the exact same settings to SW3

This is what I have set for User Script:
[image]

For the buttonlog.py file I am using the [user_func1.py test ](https://github.com/PiSupply/PiJuice/blob/master/Software/Test/user_func1.py)

If I press SW2 or SW3 I nothing happens, no log file is created in /home/pi. 

If I run python /home/pi/buttonlog.py from SSH cli the user_func1 file creates successfully.

Any ideas what I am doing wrong?

tvoverbeek, 2021-10-03:
Yes. Your script is owned by root.
The pijuice service only executes a user script when it has a non-privileged owner (i.e. not root)
and the owner belongs to the pijuice group.
The pi user already belongs to the pijuice group.
So `sudo chown pi:pi /home/pi/buttonlog.py` and it should work.

kymadn, 2021-10-04:
Thanks for sharing your expertise Ton. Your feedback is hugely appreciated. 

I can confirm buttonlog.py now works after following your advice.

However, my end goal is to open a vlc rtsp stream on a button press. 

Currently I am using a file /home/pi/edencam3.py. This file then executes edencam3.sh (maybe I can execute edencam3.sh direct from PiJuice user script, rather then a python script to execute a shell script? 

[image]

Here is a copy of edencam3.py:
`#!/usr/bin/python

import os
import sys
import logging

os.system("sh edencam3.sh")

logger = logging.getLogger('user_func1')
hdlr = logging.FileHandler('/home/pi/user_func1.log')
formatter = logging.Formatter('%(asctime)s %(levelname)s %(message)s')
hdlr.setFormatter(formatter)
logger.addHandler(hdlr) 
logger.setLevel(logging.INFO)

logger.info(str(sys.argv))
`
Here is a copy of edencam.sh:

`#!/bin/bash

sudo cvlc "rtsp://192.168.3.103/camcolor"
`
Running the command **cvlc "rtsp://192.168.3.103/camcolor"** successfully opens the video stream on my HDMI display. 

Also running: **python /home/pi/edencam.py** successfully opens the video stream on my HDMI display.

When I press SW2 or SW3 the user_func1.log is updated. So edencam3.py script runs. However, I cant work out why the video stream does not open?

kymadn, 2021-10-04:
solved

tvoverbeek, 2021-10-04:
Interested how you solved it.
Anyway user functions should be short since they are executed as part of
the pijuice service.
So if they run for a long time they block the service.

kymadn, 2021-10-05:
Hey Ton, Your suggestion got me back on track. However, I am still having some trouble.

My end goal is to open a vlc rtsp camera stream on a button press. I have several different cameras I would like to toggle between using a button press. Perhaps Pijuice user scripts are not suitable and I would be better of using Raspberry Pi GPIO's. 

Currently I am executing file /home/pi/edencam3.py from pijuice user script to start my camera stream (see below). 
```
#!/usr/bin/python

import os

os.system("cvlc 'rtsp://192.168.3.103/camcolor'")
```

I can confirm the camera stream opens as desired. However, It appears the Pi is getting hung inside edencam3.py script. I cannot send a System Halt Power Off after edencam3.py has executed and the camera stream has started.

As you said, if the script runs for a long time it will block the service. Is there an exit command I could add to the bottom of my script, which would exit the script but keep the camera stream running?

tvoverbeek, 2021-10-05:
You should run the cvlc command in the background so it does not block the service.
Look at the python docs for subprocess.Popen or (deprecated) os.spawn.