PiSupply/PiJuice #791: WakeUp alarm problem with UTC

Jan36Nikon, 2021-10-31:
Hi,

I am experiencing weird behavior with WakeUP time when I set 7;8;9;10;11 as hours to WakeUp in UTC.
Right now I have UTC+1 timezone so I am expecting to go get Wakeups on 8,9,10,11,12 in local time.

But in reality, I am getting 9,10,11,12,13 Wakeups in local time.

Did anyone know how to solve it?
[image]

Thanks, Jan

tvoverbeek, 2021-10-31:
Agter today it should be OK.
UTC does not have summer time. For me (living in Austria) summer time is UTC+2 and from today standard time UTC+1.

Jan36Nikon, 2021-10-31:
yes, I also have UTC+1 from today.

From my point it looks like it is not working correctly and pijuice is still behave like in UTC+2
Do you mean, that I will need to change the time during another time change?

Thanks

tvoverbeek, 2021-10-31:
Sorry, I did not look good enough at your screen shot which as taken today (while we both are on UTC+1).
But the reason is also in the screenshot:

The alarms are set for 7:57:20, 8:57:20, etc. UTC.
It takes some time to boot before the log message is sent.
This could correspond to 8, 9, 10 etc. UTC, giving 9, 10, 11, etc in local time.
Put both the minutes and seconds  field at 0 and see if that makes any difference (I am pretty sure it does).

tvoverbeek, 2021-11-27:
Closing the issue

closed 2021-11-27 by tvoverbeek