PiSupply/PiJuice #792: Pijuice on Openwrt with Raspberry?

OhSoGood, 2021-11-06:
Hi,

Raspberry run fine on openwrt (21.02.1). Has anyone managed to run pijuice software on it? Is there any binary available somewhere? 

@shawaj : any feedback on it? That would be so cool...

tvoverbeek, 2021-11-06:
Openwrt has python support, so it should be possible to port it to openwrt.
Do not know how the I2C support is in openwrt (needed for communication with the PiJuice).

OhSoGood, 2021-11-06:
@tvoverbeek , this is a detailed howto to activate and use I2C on a Raspberry (here a 1 model B, but should be nearly the same on other models): https://gist.github.com/David-Lor/d18b2d911a327395ed735c13f2c617b4

tvoverbeek, 2021-11-06:
I2C setup is the same as on RaspberryPi OS.
So  porting should be relatively easy.
Pay attention to different user names. The code relies on sudo no password access for the defalt user (Pi on RPi-OS)
and some utilities run under a diffferent non-privileged userid (pijuice).
I do not use openwrt, so you need somebody who knows openwrt and RaspberryPi-OS (debian) to do the port.
The current RPi code uses Debian Buster and Python 3.7