PiSupply/PiJuice #794: Boot Pi with J5/SW2 Pi Zero

GekoCH, 2021-11-11:
I have a battery powerd Setup that gets charged form time to time manualy an external power supply.
Now I think I tried every possibility but my case seems not supported by the PiJuice board.

When the PiJuice is in deep sleep mode (0.5mA) I can't wake the Pi up by pressing SW1 or SW2. I tried with or without RUN Pin.
When the PiJuice is not in deep sleep (90mA) I can wake up the Pi by pressing the SW1 or SW2 switch.

So is there no way that I can boot the Pi when I powered down the PiJuice into deep sleep mode?

GekoCH, 2021-11-12:
Ok after some more investigation I might found a bug or a feature?

I can power up the Pi Zero with the SW1/SW2 button as long as I've done one boot with a USB / Power Cable attached to the PiJuice.
Once I detach the USB Cable and do a SYS_FUNC_HALT_POW_OFF (current draw of ~0.5mA) I can boot the Pi by using the SW1/SW2 button as long as I don't detach / reattach the Battery.

So is this suppose to happen or is there a bug?

tvoverbeek, 2021-11-12:
Nothing odd.
The SW2 press has to be detected by the firmware running on the PiJuice firmware of the MCU
The MCU needs power to run. No battery and no power to the Pi and PiJuice USBs -> No dice.
Since the power needed by the MCU in deep sleep is very low only a little charge in the battery is
sufficient to get it going..

GekoCH, 2021-11-12:
but why does the MCU sometimes power the PI (so it can boot) and sometimes not after I wake the MCU by pressing SW1 or SW2?

tvoverbeek, 2021-11-12:
Assuming you are running on battery (first post says battery powered setup).
Battery needs to have sufficient charge to provide both the MCU and the Pi with power (5V via GPIO).
The Pi needs an order of magnitude more power than the MCU.
My guess is you are on the limit with your battery or your battery is getting bad.

GekoCH, 2021-11-12:
battery can't be the limiting factor, Its a 5400mAh, lipo with 3.9V. Wires to the battery are also sufficient enough to handle "high current".
But generally is the PiJuice able to run on battery alone and boot /start the Pi once the SW1 is pushed without needing additional power from a USB charger etc.?

tvoverbeek, 2021-11-12:
In general yes.
But it also depends on the total load (external peripherals attached to the Pi).
E.g SSDs can have a high inrush current. If the demanded power is too much the 5V to the Pi drops and the Pi stops.

GekoCH, 2021-11-13:
ok the additional load (at the USB of the Pi) is fairly low at 0.5A during startup of the pi so the PiJuice should be able to handle this.
It seems now pretty stable. The last 20 boots from the battery were successful, however the Fuel gauge IC still delivers strange value but maybe it needs some time to "learn"...