PiSupply/PiJuice #795: Battery and Temp Issue

miro-be, 2021-11-25:
Hi,

We have designed an equipment which integrates 2 hat on a raspbery, including the pijuice on the 1st floor.
We observe several battery swelling and a high temperature of almost constant 53 ° degrees.

it looks like the battery is still charging and never reaches 100%, is it normal ?

Do you have any tips or advice to resolve this swelling problem? We have a few ideas like changing the location of the pijiuice card and putting it on the top floor so that it is no longer sandwiched between 2 hot cards but I don't know if that will be enough?

the box is generally always connected to the power supply and only uses the battery in the event of a power failure.
is this use appropriate for your product?

My second issue is when I upgraded pijuice base 1.6, I don't see the change in the firmware menu , it's still written 1.4, is this normal ?

I am attaching some pictures of our case.

Regards,
[image]
[image]
[image]
[image]
[image]

tvoverbeek, 2021-11-25:
Battery swelling is bad and could end with an exploding battery.
You definitely need a cooler place for the battery.
Battery not reaching 100% when running is normal. The charge level is estimated and something in the 9x% range is normal.
I would definitely place the PiJuice on top and well ventilated. I realise you need to change the case for this.
When the battery is too hot it will not charge for safety reasons (unless you fiddled with the battery profile)
Alternatively you could place only the battery (a different one) in a cooler place

miro-be, 2021-12-24:
Thanks for the  very quick reply.

We will study for a new case and create an independant place for the battery 

What kind of battery do you recommend for use  in a constant charge of a pijuice ?

Do you think that once in a cool place the battery will stop being in constant charge and no longer swell? 

I have heard that Lithium Ion is not hightly recomended for this case.