PiSupply/PiJuice #797: Original PiJuice Battery weather endurance

Jan36Nikon, 2021-12-06:
Good morning,

I want to ask for your experience with original PiJuice battery in cold weather like -10°C.

Right now I am testing the capacity of PiJuice battery 2500 mAh in temperatures around -2°C and it is looking more than OK. After one week it lost only 10% of capacity while switching on only 3 times per day everytime for 3 minutes.

Do you have any experience with the cold weather influence on battery?

Is it OK to let it stay in freezing temperature?

Battery will be charged only manually when temperature increases to not freezing degrees.

Thanks a lot,
Jan

pedrosk, 2021-12-07:
Jan,

My PI4 with BX7 is outside (today is -13C). It is connected to a 12V battery and 20W solar panel. I can tell you that that is not enough to run it for more than 2 days. If I let the PI and PiJuice run on its own I get about 70 minutes out of the set up. 

I have it in an enclosure and I measure the ambient temperature in the enclosure. With -16C outside the temperature in the box is nearing zero 0C.

Note that the cold temp on the BX7 battery should not drop bellow zero. 

Here is the last 24 hours or ambient temperatures. It was -8C last night 

Last 36hours temperatures and humidity in the box with the PI and PIjuice
[image]

Temperature in the box right now with -15C outside. I can update how did the temperature in the enclosrure with help in the morning (if the 12V will last)

[image]

Also note that the PiJuice is reporting 20C 
[image]

I did not script up the harvesting of data from PiJuice yet, but you could use the example test included in this repo and write your own python code to do that. (that is my plan)

Let me know if this helped

Ed

Jan36Nikon, 2021-12-07:
Thank you for your tips.
In my case device is turned off most of the time.
It switch ON only 3 times per day for 3 minutes each time.

In this case it looks like original PiJuice 2500mAh battery is able to stay without charging for 50 days.

But I can be sure that battery itself is exposed to freezing temperatures during this winter.
It looks like capacity is not dropping down too much.

But my question is if the PiJuice battery can be in freezing temperature for a long time. Of course without charging.

Jan

pedrosk, 2021-12-07:
@Jan36Nikon 

You probably have this battery:
```

(5) PJBP7X_1600

    Capacity: 1600 mAh,
    Charge current: 925 mA,
    Termination current: 50 mA,
    Regulation voltage: 4180 mV,
    Cut-off voltage: 3000 mV,
    Cold temperature: 0 °C,
    Cool temperature: 2 °C,
    Warm temperature: 50 °C,
    Hot temperature: 70 °C,
    NTC B constant: 3380 K,
    NTC resistance 10000 Ω.
```

So if you look above it says the cold temperature is Zero C (freezing).  From what I was reading over time the cold can do permanent damage. There is some phenomenon where the battery will create some crystals and it will lower the capacity permanently. There is a process to over come this and dissolve them with sofisticated process and reverse the flow. I don't think it would be possible on these batteries and I think it would probably make them explode if not done right. That same thing can happen I think if the batteries do not discharge often enough and deep enough. I think a good resource would be ask solar organizations that use these batteries, or Universities. 

I am trying to shelter the unit as much as possible. 

On a different note: I operate about 100 wildlife cameras that run Nickel Metal Hydrid rechargeable batteries. It gets cold here and hot in the summer. In winter the batteries don't last long and need to be swapped often, on the the other hand there were cameras that run on the same batteries that were swapped at the end of February till Jun, July, August, September and even End of November.... Not sure if LiPo can take the same abuse. 

One thing seems to true that if the PiRuns the battery stays warm. Even the enclose stayed at 0C while it was -15C outside for over 13hrs. So maybe keep it running and figure out how to charge the system constantly with a larger battery and panels? Don't know.

Here is my setup:
There is an old retired hive box that is holding all the gear, and this is all in a shelter with hay that protects it from weather a little

[image]
than the hive protects even more:

and the gray boxes 
[image]
house the electronic, they have a temp sensor there as well:
[image]

If I may ask where, what country are you using these ? What is the application?

Jan36Nikon, 2021-12-07:
100 cameras. Great. You should have fantastic photos.
My camera will be in Czech republic.
Still testing right now. Outdoor temperature is around -2 °C right now and temp in the box is around 0 °C.

It will be nice to get some info from other users experience according to freezing temperature.

Thanks, have a nice day,
Jan

pedrosk, 2021-12-07:
I just ordered the 12000 battery. Arrived yesterday so should have more data as the project goes on. At the moment I am also running on the stock battery. 

I am in Wisconsin. But If my outdated memory does not fail too well. Czech republic used NOT have the cold as we get up here so maybe you will be in less risk? Would be interesting to add an ambient temperature reading and store it the log along with PiJuice data. 
Quick question. When your camera will run, it is just going to be PIR activated or by the code and will run for a few minutes?

Jan36Nikon, 2021-12-07:
It could be -15 °C or more but not for a long time.

This article is interesting:
https://drones.stackexchange.com/questions/378/do-low-temperatures-damage-lipo-batteries

Have a nice day,
Jan

pedrosk, 2021-12-07:
@Jan36Nikon I will try to find the link. I read a research paper about two or three years ago something about a bellow freezing battery should not be charged. I don't remember the exact threshold but they stated that because the electrolyte is not a fluid enough the charging a frozen battery can cause damage. If I remember they stated something above +5 or little more to charge. But I could be wrong. 

Zdravim Ed

tvoverbeek, 2021-12-07:
The article @Jan36Nikon is a good summary. Includes everything you should
know.

pedrosk, 2021-12-07:
@tvoverbeek I think the heat is a bigger problem and some cooling would be needed if installed potentially outside. 
Can or should a heatsink with a fan be screwed on top (like it used to be in the old Pentium MMX series) a good idea? Maybe liquid cooling. The measurements seems to be in favor in the cold weather, but I sense trouble once it gets hot....
Or mounting anything on top battery to decipate the heat would be too dangerous?

ed